foxglove/ros-foxglove-bridge #2: release ros-foxglove-bridge c++ ros2 node

defunctzombie, 2022-11-01:


jhurliman, 2022-11-21:
ROS 2 humble bloom has started: https://github.com/ros/rosdistro/pull/35378

defunctzombie, 2022-11-21:
🌱

jhurliman, 2022-11-22:
The blooms are blooming

defunctzombie, 2022-11-22:
@jhurliman does the closure of this issue mean the package is released into the ros2 testing apt repos?

defunctzombie, 2022-11-22:
Re-opening. Let's close this when the referenced PR is merged so we can use this to track that PR.

jhurliman, 2022-11-22:
The closure of this issue means all of the ROS 2 PRs have merged except for the Humble PR, since Humble is mid-release so it's been tagged as "held for sync". It will be merged when the sync is complete.

@defunctzombie I'm not sure what your plan is for this ticket so I've assigned you as owner. You can close it out at the milestone of your choosing (upstream PR marked held for sync, upstream PR merged, distro sync begins, apt package can be installed on your local machine).

defunctzombie, 2022-11-22:
I am following what was done in this ticket: https://github.com/foxglove/ros-foxglove-bridge/issues/1 which closed when the linked PRs for the supported distros closed. I am assigning this back to you to follow-through on making sure the relevant PRs merge for the releases we are interested in.

closed 2022-11-28 by jhurliman