foxglove/ros-foxglove-bridge #13: [RFC] MCAP WebSocket protocol

jhurliman, 2022-11-08:
MCAP WebSocket
==============

JSON-encoded control messages in ws-protocol are converted to MCAP Metadata records. All messages are binary. MCAP native record types are used for Schemas, Channels, and Messages. MCAP records are sent as-defined including the size prefix, even though it is often redundant with the WebSocket framing. This allows fragmented delivery of large messages or attachments.

### Server Messages
serverInfo { name, capabilities } -> Metadata { op: "server_info", name, capabilities }

### Client Messages
subscribe { [subscriptions]: { id, channelId } } -> Metadata { op: "subscribe", subscription_id, channel_id }
unsubscribe { [subscriptionIds] } -> Metadata { op: "unsubscribe", subscription_id }
... -> Metadata { op: "client_info", name, capabilities }

### Common Messages
advertise { [channels] } -> Schema, Channel
unadvertise { [channelIds] } -> Metadata { op: "unadvertise", channel_id }
message { timestamp, subscriptionId, data } -> Message
status { level, message } -> Metadata { op: "status", level, message }

TBD:
- Parameters
- Services
- Actions

jhurliman, 2022-11-08:
> Including client-published messages? In https://github.com/foxglove/ws-protocol/pull/250 we have a spec update that uses JSON for client-published messages.

My thinking is to always use MCAP as the serialization format. It's valid to publish JSON messages, but from a WebSocket transport point-of-view they would be binary payloads.

jhurliman, 2022-11-08:
> This alone would allow fragmentation but still requires all fragments to be delivered in order with no interruption. Will there be a way to tag fragmented messages with an id so they can be interspersed with other smaller messages?

The current protocol is based purely on MCAP records, and I'm reticent to drift away from that since it has the nice property of being able to stream from a file for DVR-like playback or stream to a file for efficient recording. It has the limitation of being unable to interleave individual records (same as the MCAP file format). My own two cents is this is acceptable for mcap-websocket and something that goes away with mcap-webtransport where every channel has its own QUIC channel.

amacneil, 2022-11-08:
I think generally this is a sensible direction to go in. We should discuss whether now is the right time, or whether there are more tactical changes we should make first.

A similar analogy is Apache Arrow (columnar table format) vs [Arrow Flight](https://arrow.apache.org/blog/2019/10/13/introducing-arrow-flight/) (gRPC API for streaming Arrow tables).

We have heard feedback that for customers with a custom robotics stack, implementing both mcap for recording and fg-websocket for live viz is duplicative (and the protocols are unnecessarily divergent).

We also know that we have features on the horizon, such as:
- DVR (alternate between live and recorded data in the same connection)
- WebTransport / WebRTC support (lossy streaming over low bandwidth links)

It would be excellent if we could retain the property that the on-wire format and the file format are the same, i.e. I could pipe all data received on a socket to a file, and that mcap file would be perfectly legitimate to open in Foxglove Studio. Likewise, I could pipe a file directly into a socket, and while the other end might be surprised to receive some messages on channels it had not subscribed to, it would be a legitimate series of packets.

[Here](https://docs.google.com/document/d/1f7ZROgjJCs060qzKPHikcWYJ7Mu6KBeBp2VLJZP-AVg/edit) is a (non-public) document I wrote about 6 months ago with some similar ideas in it.

jtbandes, 2022-11-08:
For posterity:

[image]

amacneil, 2022-11-09:
I thought some more about this.

I feel like there are a lot of unanswered questions about how this protocol will evolve (e.g. how are we going to solve ROS concepts like parameters/services/actions, how might we solve DVR, support different underlying transport protocols, etc). Therefore, if we want to go down this path, we are going to need a long baking time before we are ready to publish it as part of the official MCAP spec, during which time we may want to break backwards compatibility (with older servers, not the existing MCAPv1 spec).

Unless I'm overthinking this, I think that is at odds with shipping a quick solution for ROS better performance ROS 2 connections.

jtbandes, 2022-11-10:
Decision for now was to move forward with more incremental updates to the foxglove websocket protocol for initial release of foxbridge.

jhurliman, 2022-11-11:
Closing for now, we can reevaluate after reaching feature parity with rosbridge

closed 2022-11-11 by jhurliman