foxglove/ros-foxglove-bridge #37: Retry loop on startup ("retry_startup_delay")

jhurliman, 2022-11-11:
Right now, the server will attempt to bind a listening socket once and fail out if it doesn't succeed. rosbridge runs an indefinite loop until the listen() succeeds or the node exits, retrying with a `sleep(retry_startup_delay)` in between. We should emulate this behavior.

achim-k, 2022-11-21:
I think we should not do anything here and keep the current behavior. The user can use existing ROS functionality to respawn the node if it crashes:

```xml
<node pkg="foxglove_bridge" ...  respawn="true" respawn_delay="5" />
```

Reference:
- ROS1: https://wiki.ros.org/roslaunch/XML/node
- ROS2: https://github.com/ros2/launch/blob/6783ca8988b3ee1910e3eb6de7388407c8c0ce0d/launch/launch/actions/execute_process.py#L230-L232

jhurliman, 2022-11-21:
As long as the node/nodelet crashes on failure to bind() I agree. I just want to make sure we don't leave the WebSocket server in a disconnected state with the node running and no way to start it.

closed 2022-11-21 by jhurliman