foxglove/ros-foxglove-bridge #38: Publish "client_count" topic

jhurliman, 2022-11-11:
Taken from rosbridge:

```
# Publisher for number of connected clients
# QoS profile with transient local durability (latched topic in ROS 1).
client_count_qos_profile = QoSProfile(
    depth=10,
    durability=DurabilityPolicy.TRANSIENT_LOCAL,
)

RosbridgeWebSocket.client_count_pub = self.create_publisher(
    Int32, "client_count", qos_profile=client_count_qos_profile
)
RosbridgeWebSocket.client_count_pub.publish(Int32(data=0))
```

achim-k, 2022-11-21:
The main motivation behind the `client_count` topic was to let a client see that they might not be the only one that is making changes to the server configuration (e.g. through setting parameters or calling other kind of ros services).

Having this is useful, but I wonder if we should not add it directly to the [ws-protocol spec](https://github.com/foxglove/ws-protocol/blob/main/docs/spec.md) instead of having it a separate ROS topic

jhurliman, 2022-11-21:
One side effect of publishing a ROS topic is that `/client_count` can end up in a recording. I'm not sure if that is useful though, it does seem like something only WebSocket-connected clients would care about.

To support this in ws-protocol, we would need something like the `status` op that servers can push to clients at will, but with more structure perhaps like a key/value pair and using `client_count` as the key?

achim-k, 2022-11-21:
> To support this in ws-protocol, we would need something like the `status` op that servers can push to clients at will, but with more structure perhaps like a key/value pair and using `client_count` as the key?

Or maybe we just add `client_count` as an optional field to the `status` message?

ndmmdn, 2025-07-17:
I was just looking for this feature, basically we have limited bandwidth on our robots in the field and we would like to use the bandwidth for other things when no foxglove client is connected, but if a client is connected we pause the background tasks to enable more responsive foxglove dashboards