foxglove/ros-foxglove-bridge #81: Debian buster build is failing in ROS buildfarm, no ros-noetic-ros-babel-fish package on that platform

jhurliman, 2022-11-29:
https://build.ros.org/job/Ndev_db__foxglove_bridge__debian_buster_amd64/11/

```
KeyError: "The cache has no package named 'ros-noetic-ros-babel-fish'"
```

achim-k, 2022-11-29:
I don't understand why, the package is available on packages.ros.org. Also on the appropriate docker image I can find that package:

`docker run -it --rm ros:noetic-ros-core-buster bash -c "apt update && apt-cache policy ros-noetic-ros-babel-fish"`

```
[...]
ros-noetic-ros-babel-fish:
  Installed: (none)
  Candidate: 0.9.3-1buster.20220926.194426
  Version table:
     0.9.3-1buster.20220926.194426 500
        500 http://packages.ros.org/ros/ubuntu buster/main amd64 Packages
```

achim-k, 2022-12-05:
This has been fixed by re-releasing `rosbridge_suite` for noetic (see https://github.com/ros/rosdistro/pull/35528)

> In a previous release my bloom had accidentally set [catkin and ros-message-runtime instead of ros-noetic-catkin and ros-noetic-message-runtime](https://github.com/RobotWebTools-release/rosbridge_suite-release/compare/06596b2e7ed9421c183b53263198a9221d4f0d06...9b8d9d107c189bdd9d238c88aff4697f9fe78284#diff-2a13f1344504379e877324d3a0375adcbcb4cb27d7e575ad8f4618a52d6a00e0R5) for debian/noetic/buster/*

This also fixed the build for `ros_babel_fish` as it has a dependency on `rosapi` (released with `rosbridge_suite`)

closed 2022-12-05 by achim-k