foxglove/ros-foxglove-bridge #84: [ros2] Client does not receive binary messages anymore after previous client disconnected

achim-k, 2022-11-29:
**Description**
When subscribing to a channel with a client and then disconnecting, the next client that subscribes to the same channel does not receive binary messages anymore

- Version: Latest (https://github.com/foxglove/ros-foxglove-bridge/commit/54595cb9c4eedf1671376d82b149c54bfba7fc63)
- Platform: Ubuntu 20.04.5 LTS (devcontainer)

**Steps To Reproduce**
The following instructions require [websocat](https://github.com/vi/websocat)

1. Publish a ros topic: `ros2 topic pub /foo std_msgs/String data:\ "hello"`
2. Launch the foxglove bridge: `ros2_foxglove_bridge`
3. Connect a client and subscribe to the channel that corresponds to the topic published in 1): `echo '{"op": "subscribe", "subscriptions": [ { "id": 0, "channelId": 3 } ]}' | websocat ws://localhost:8765 --protocol foxglove.websocket.v1 --no-close` (channel ID might be different)
4. Verify that the client receives binary messages before disconnecting (ctrl+c)
5. Connect a new client by re-running the command from 3)
6. Observe that no binary messages are received

**Expected Behavior**
The new client should receive binary messages when subscribing

**Actual Behavior**
The new client does not receive binary messages when a previous client that subscribed to the same channel has disconnected before

achim-k, 2022-11-29:
Interesting finding: I observe that the second client starts receiving binary messages again after I execute `ros2 topic echo /foo`. Only this extra subscription triggers the subscriber in foxglove_bridge to receive messages again :thinking:

achim-k, 2022-11-29:
The problem is not related to client disconnections. It can be reproduced within a single client section by subscribing, unsubscribing and resubscribing:
```
{"op": "subscribe", "subscriptions": [ { "id": 0, "channelId": 3 } ]}
G���+,hello
PMn++,hello
I�g+,hello
{"op": "unsubscribe", "subscriptionIds": [0] } 
{"op": "subscribe", "subscriptions": [ { "id": 0, "channelId": 3 } ]}
# No messages received anymore
```

jhurliman, 2022-11-30:
It looks like something to do with the ROS 2 executor system, either something we don't understand or a bug. Adding this to the ROS 2 node fixes the issue:

```cpp
_updateTimer = this->create_wall_timer(100ms, []() {});
```

You can use any frequency. On the third subscription, the transient-local message won't be replayed until the timer executes (you can confirm this by setting a slower frequency and looking at timestamps, and/or adding more logging).

jhurliman, 2022-11-30:
Fixed in https://github.com/foxglove/ros-foxglove-bridge/pull/77 although we don't understand why.

closed 2022-11-30 by jhurliman