foxglove/ros-foxglove-bridge #89: Move C++ websocket server code back to ws-protocol repo

jtbandes, 2022-11-30:
Since this bridge makes use of the Foxglove WebSocket protocol, the generic C++ server implementation should live with the rest of the code at https://github.com/foxglove/ws-protocol. Unless there is a particular reason to have two server implementations, we should probably replace the old one with this one.

jhurliman, 2022-11-30:
I suggest moving this issue to https://github.com/foxglove/ws-protocol since the result would be a PR to that repo and no change to this repo.

jtbandes, 2022-11-30:
I'd rather not have two copies of the code and have to keep them in sync, so the result should also include deleting the version in this repo.

amacneil, 2022-12-20:
It sounds like time for a monorepo

[image]

jlack1987, 2024-09-08:
Which implementation of the websocket protocol is most up-to-date/newer? It seems [this one](https://github.com/foxglove/ws-protocol) has more recent commits than this repo no?

achim-k, 2024-09-11:
> Which implementation of the websocket protocol is most up-to-date/newer? It seems [this one](https://github.com/foxglove/ws-protocol) has more recent commits than this repo no?

This repo basically uses a copy of the C++ server code in https://github.com/foxglove/ws-protocol, with only a few minor differences.

I'm not sure if there is an elegant way to add `ws-protocol` as a build dependency remove the server code in this pacakge. Maybe we could do so using [FetchContent](https://cmake.org/cmake/help/latest/module/FetchContent.html)?

jlack1987, 2024-09-11:
Yeah it's doable via the `FetchContent` stuff in cmake. I actually did just that on that repo a few days ago. That's when I ran into this repo as well and got confused. Some minor cmake mods are needed though on `ws-protocol` to ensure some of `websocketpp`'s dependencies are available(zlib, openssl, and boost iirc). Very easy stuff to do though

closed 2025-02-04 by amacneil

amacneil, 2025-02-04:
We're currently working on a new C++ SDK, after which we will update `ros-foxglove-bridge` to use the new SDK, and figure out shared libraries at that time.