foxglove/ros-foxglove-bridge #94: Subscription smoke test is failing on Rolling Jammy in ROS buildfarm

jhurliman, 2022-11-30:
See <https://build.ros2.org/job/Rdev__foxglove_bridge__ubuntu_jammy_amd64/21/>.

Connection and publishing tests are working, but no messages are received in the subscription test (this is different from the callback_group issue since no message is ever received). This test is passing for Humble Jammy; see <https://build.ros2.org/job/Hdev__foxglove_bridge__ubuntu_jammy_amd64/9/>.

```
2: [ RUN      ] SmokeTest.testSubscription
2: [INFO] [1669838650.865567189] [foxglove_bridge]: [WS] Client 127.0.0.1:43488 connected via /
2: [INFO] [1669838651.349482682] [foxglove_bridge]: Subscribing to topic "/pub_topic" (std_msgs/msg/String) on channel 3
2: /tmp/ws/src/foxglove_bridge/ros2_foxglove_bridge/tests/smoke_test.cpp:76: Failure
2: Expected equality of these values:
2:   std::future_status::ready
2:     Which is: 4-byte object <00-00 00-00>
2:   msgFuture.wait_for(std::chrono::seconds(5))
2:     Which is: 4-byte object <01-00 00-00>
2: [INFO] [1669838656.349140779] [foxglove_bridge]: [WS] Client 127.0.0.1:43488 disconnected
2: [INFO] [1669838656.350879868] [foxglove_bridge]: Unsubscribing from topic "/pub_topic" (std_msgs/msg/String) on channel 3
2: [  FAILED  ] SmokeTest.testSubscription (5502 ms)
```

achim-k, 2022-11-30:
It looks like the test even crashed for the previous runs:

- https://build.ros2.org/job/Rdev__foxglove_bridge__ubuntu_jammy_amd64/20/testReport/(root)/projectroot/smoke_test/
- https://build.ros2.org/job/Rdev__foxglove_bridge__ubuntu_jammy_amd64/19/testReport/(root)/projectroot/smoke_test/

achim-k, 2022-12-16:
Very likely related to #120

PR #122 (2022-12-19): Use a single reentrant callback group for all subscriptions
**Public-Facing Changes**
- Fix subscriptions not working on rolling

**Description**
- Callback groups that are created after the executor is already spinning do not work correctly, see https://github.com/ros2/rclcpp/issues/2067
- Instead of dynamically creating callback groups for each advertised channel, we now use a single reentrant callback group for all channels
  - Consequence is, that message callbacks of the same topic (e.g. when having multiple clients) can now be called in parallel. However, this is fine since the `server:: [... cut]
commit 3ffb02bd (2022-12-19): Use a single reentrant callback group for all subscriptions (#122)

**Public-Facing Changes**
- Fix subscriptions not working on rolling

**Description**
- Callback groups that are created after the executor is already
spinning do not work correctly, see
https://github.com/ros2/rclcpp/issues/2067
- Instead of dynamically creating callback groups for each advertised
channel, we now use a single reentrant callback group for all channels
- Consequence is, that message callbacks of the same topic (e.g. when
having multiple clients) can now be called in parallel. However, this is
fine since the `se [... cut]
closed 2022-12-19 by achim-k