foxglove/ros-foxglove-bridge #120: Subscriptions not working on ROS2 rolling

achim-k, 2022-12-16:
**Description**

- Version: latest
- Platform: `ROS2 rolling / ubuntu jammy` (provided docker image)

**Steps To Reproduce**
1. Run the bridge e.g. with `docker run --pull --rm -it --name foxbridge --net=host ghcr.io/foxglove/rolling-ros2-bridge`
2. Publish a periodic topic e.g. `docker exec -it foxbridge bash -c "source /opt/ros/rolling/setup.bash && ros2 topic pub /foo std_msgs/String data:\ 'hello'"`
3. Connect a client and observe that subscriptions are not working
  - E.g. use Studio with a raw panel (https://studio.foxglove.dev/?ds=foxglove-websocket&ds.url=ws%3A%2F%2Flocalhost%3A8765)

**Expected Behavior**
Client should receive messages of subscribed channels

defunctzombie, 2022-12-16:
https://linear.app/foxglove/issue/FG-1034/subscriptions-not-working-on-ros2-rolling-[foxgloveros-foxglove-bridge

Don't know why the bot did not comment on this.

achim-k, 2022-12-16:
Observation: Making a graph change by e.g. publishing on a random topic (ros2 topic pub /asdf std_msgs/String data:\ "meh") triggers the bridge to send messages to the client :thinking:

achim-k, 2022-12-16:
Underlying rclcpp issue: https://github.com/ros2/rclcpp/issues/2067

PR #122 (2022-12-19): Use a single reentrant callback group for all subscriptions
**Public-Facing Changes**
- Fix subscriptions not working on rolling

**Description**
- Callback groups that are created after the executor is already spinning do not work correctly, see https://github.com/ros2/rclcpp/issues/2067
- Instead of dynamically creating callback groups for each advertised channel, we now use a single reentrant callback group for all channels
  - Consequence is, that message callbacks of the same topic (e.g. when having multiple clients) can now be called in parallel. However, this is fine since the `server:: [... cut]
commit 3ffb02bd (2022-12-19): Use a single reentrant callback group for all subscriptions (#122)

**Public-Facing Changes**
- Fix subscriptions not working on rolling

**Description**
- Callback groups that are created after the executor is already
spinning do not work correctly, see
https://github.com/ros2/rclcpp/issues/2067
- Instead of dynamically creating callback groups for each advertised
channel, we now use a single reentrant callback group for all channels
- Consequence is, that message callbacks of the same topic (e.g. when
having multiple clients) can now be called in parallel. However, this is
fine since the `se [... cut]
closed 2022-12-19 by achim-k