foxglove/ros-foxglove-bridge #147: Consider removing comments from message schemas

achim-k, 2023-02-07:
**Description**
When advertising channels, we currently send the full message schema including comments to clients. This can be rather verbose when inspecting the websocket traffic:

<details>
<summary>With comments (4828 characters)</summary>
<i>
{"channels":[{"encoding":"cdr","id":2,"schema":"# This message contains a parameter event.\n# Because the parameter event was an atomic update, a specific parameter name\n# can only be in one of the three sets.\n\n# The time stamp when this parameter event occurred.\nbuiltin_interfaces/Time stamp\n\n# Fully qualified ROS path to node.\nstring node\n\n# New parameters that have been set for this node.\nParameter[] new_parameters\n\n# Parameters that have been changed during this event.\nParameter[] changed_parameters\n\n# Parameters that have been deleted during this event.\nParameter[] deleted_parameters\n\n================================================================================\nMSG: builtin_interfaces/Time\n# This message communicates ROS Time defined here:\n# https://design.ros2.org/articles/clock_and_time.html\n\n# The seconds component, valid over all int32 values.\nint32 sec\n\n# The nanoseconds component, valid in the range [0, 10e9).\nuint32 nanosec\n\n================================================================================\nMSG: rcl_interfaces/Parameter\n# This is the message to communicate a parameter. It is an open struct with an enum in\n# the descriptor to select which value is active.\n\n# The full name of the parameter.\nstring name\n\n# The parameter's value which can be one of several types, see\n# `ParameterValue.msg` and `ParameterType.msg`.\nParameterValue value\n\n================================================================================\nMSG: rcl_interfaces/ParameterValue\n# Used to determine which of the next *_value fields are set.\n# ParameterType.PARAMETER_NOT_SET indicates that the parameter was not set\n# (if gotten) or is uninitialized.\n# Values are enumerated in `ParameterType.msg`.\n\n# The type of this parameter, which corresponds to the appropriate field below.\nuint8 type\n\n# \"Variant\" style storage of the parameter value. Only the value corresponding\n# the type field will have valid information.\n\n# Boolean value, can be either true or false.\nbool bool_value\n\n# Integer value ranging from -9,223,372,036,854,775,808 to\n# 9,223,372,036,854,775,807.\nint64 integer_value\n\n# A double precision floating point value following IEEE 754.\nfloat64 double_value\n\n# A textual value with no practical length limit.\nstring string_value\n\n# An array of bytes, used for non-textual information.\nbyte[] byte_array_value\n\n# An array of boolean values.\nbool[] bool_array_value\n\n# An array of 64-bit integer values.\nint64[] integer_array_value\n\n# An array of 64-bit floating point values.\nfloat64[] double_array_value\n\n# An array of string values.\nstring[] string_array_value\n","schemaName":"rcl_interfaces/msg/ParameterEvent","topic":"/parameter_events"},{"encoding":"cdr","id":1,"schema":"##\n## Severity level constants\n## \n## These logging levels follow the Python Standard\n## https://docs.python.org/3/library/logging.html#logging-levels\n## And are implemented in rcutils as well\n## https://github.com/ros2/rcutils/blob/35f29850064e0c33a4063cbc947ebbfeada11dba/include/rcutils/logging.h#L164-L172\n## This leaves space for other standard logging levels to be inserted in the middle in the future,\n## as well as custom user defined levels.\n## Since there are several other logging enumeration standard for different implementations,\n## other logging implementations may need to provide level mappings to match their internal implementations.\n##\n\n# Debug is for pedantic information, which is useful when debugging issues.\nbyte DEBUG=10\n\n# Info is the standard informational level and is used to report expected\n# information.\nbyte INFO=20\n\n# Warning is for information that may potentially cause issues or possibly unexpected\n# behavior.\nbyte WARN=30\n\n# Error is for information that this node cannot resolve.\nbyte ERROR=40\n\n# Information about a impending node shutdown.\nbyte FATAL=50\n\n##\n## Fields\n##\n\n# Timestamp when this message was generated by the node.\nbuiltin_interfaces/Time stamp\n\n# Corresponding log level, see above definitions.\nuint8 level\n\n# The name representing the logger this message came from.\nstring name\n\n# The full log message.\nstring msg\n\n# The file the message came from.\nstring file\n\n# The function the message came from.\nstring function\n\n# The line in the file the message came from.\nuint32 line\n\n================================================================================\nMSG: builtin_interfaces/Time\n# This message communicates ROS Time defined here:\n# https://design.ros2.org/articles/clock_and_time.html\n\n# The seconds component, valid over all int32 values.\nint32 sec\n\n# The nanoseconds component, valid in the range [0, 10e9).\nuint32 nanosec\n","schemaName":"rcl_interfaces/msg/Log","topic":"/rosout"}],"op":"advertise"}
</i>
</details>

<details>
<summary>Without comments (1414 characters)</summary>
<i>
{"channels":[{"encoding":"cdr","id":2,"schema":"\nbuiltin_interfaces/Time stamp\n\nstring node\n\nParameter[] new_parameters\n\nParameter[] changed_parameters\n\nParameter[] deleted_parameters\n\n================================================================================\nMSG: builtin_interfaces/Time\n\nint32 sec\n\nuint32 nanosec\n\n================================================================================\nMSG: rcl_interfaces/Parameter\n\nstring name\n\nParameterValue value\n\n================================================================================\nMSG: rcl_interfaces/ParameterValue\n\nuint8 type\n\n\nbool bool_value\n\nint64 integer_value\n\nfloat64 double_value\n\nstring string_value\n\nbyte[] byte_array_value\n\nbool[] bool_array_value\n\nint64[] integer_array_value\n\nfloat64[] double_array_value\n\nstring[] string_array_value\n","schemaName":"rcl_interfaces/msg/ParameterEvent","topic":"/parameter_events"},{"encoding":"cdr","id":1,"schema":"\nbyte DEBUG=10\n\nbyte INFO=20\n\nbyte WARN=30\n\nbyte ERROR=40\n\nbyte FATAL=50\n\n\nbuiltin_interfaces/Time stamp\n\nuint8 level\n\nstring name\n\nstring msg\n\nstring file\n\nstring function\n\nuint32 line\n\n================================================================================\nMSG: builtin_interfaces/Time\n\nint32 sec\n\nuint32 nanosec\n","schemaName":"rcl_interfaces/msg/Log","topic":"/rosout"}],"op":"advertise"}
</i>
</details>

The above examples were recorded when the bridge was launched without any other nodes. I guess the difference will be much larger when e.g. custom message types are used

jtbandes, 2023-02-13:
> This can be rather verbose when inspecting the websocket traffic

I don't think this is a sufficient motivation to remove the comments. Most users do not inspect websocket traffic, they just use the bridge and expect it to work.

achim-k, 2023-02-13:
Ok I'm gonna close this issue. It's also less of an issue when #146 was addressed

closed 2023-02-13 by achim-k