foxglove/ros-foxglove-bridge #150: ros1 docker image is broken

jhurliman, 2023-02-07:
```
$ docker run --rm -it -v /opt/ros:/opt/ros --net=host ghcr.io/foxglove/noetic-ros1-bridge
/ros_entrypoint.sh: line 5: /opt/ros/noetic/setup.bash: No such file or directory
```

achim-k, 2023-02-07:
This happens when you don't have a local ros installation (`/opt/ros/...`). Leaving out the  `-v /opt/ros:/opt/ros` makes it work. It's the same for #149

I think we should consider removing documentation for the docker images. They offer little value as they don't work with custom types.

defunctzombie, 2023-02-09:
I second getting rid of these images. I do not think they fall into a typical ROS workflow. It is not worth the time to maintain and fix them.

PR #159 (2023-02-13): Do not build docker images, remove corresponding documentation
**Public-Facing Changes**
- Disable building & publishing of docker images

**Description**
We have decided to not build & publish docker images anymore, as they offer limited value and do not work with custom message types.

Fixes #149 ([FG-1801](https://linear.app/foxglove/issue/FG-1801/ros2-docker-image-is-broken-[foxgloveros-foxglove-bridge-149]))
Fixes #150 ([FG-1802](https://linear.app/foxglove/issue/FG-1802/ros1-docker-image-is-broken-[foxgloveros-foxglove-bridge-150]))
commit 31805d2e (2023-02-13): Do not build docker images, remove corresponding documentation (#159)

**Public-Facing Changes**
- Disable building & publishing of docker images

**Description**
We have decided to not build & publish docker images anymore, as they
offer limited value and do not work with custom message types.

Fixes #149
([FG-1801](https://linear.app/foxglove/issue/FG-1801/ros2-docker-image-is-broken-[foxgloveros-foxglove-bridge-149]))
Fixes #150
([FG-1802](https://linear.app/foxglove/issue/FG-1802/ros1-docker-image-is-broken-[foxgloveros-foxglove-bridge-150]))
closed 2023-02-13 by achim-k