foxglove/ros-foxglove-bridge #157: Docs are a little sparse on configuration

rgov, 2023-02-09:
https://foxglove.dev/docs/studio/connection/using-foxglove-bridge - This page tells you how to install it and which config options it takes, but it doesn't have an example of how to launch it.

https://github.com/foxglove/ros-foxglove-bridge#running-the-bridge - explains how to launch it but doesn't show how to set configuration options on it.

For completeness it would be nice to have some examples like these, to spare the user having to look up the syntax roslaunch uses:

```
$ roslaunch foxglove_bridge foxglove_bridge.launch port:=1989 --screen
```

```xml
<launch>

    <!-- Including in another launch file --> 
    <include file="$(find foxglove_bridge)/launch/foxglove_bridge.launch">
        <arg name="port" value=1989 />
    </include>

</launch>
```

rgov, 2023-02-10:
The `topic_whitelist` parameter is also a list which is a little weird to pass:

```
<arg name="topic_whitelist" value="[&quot;^(?!/camera/image$).*&quot;]" />
```

achim-k, 2023-02-10:
> The `topic_whitelist` parameter is also a list which is a little weird to pass:
> 
> ```
> <arg name="topic_whitelist" value="[&quot;^(?!/camera/image$).*&quot;]" />
> ```

Agreed, it's not very intuitive.
Maybe the best for you would be to create your own version of `foxglove_bridge.launch` so you can use `<rosparam>`:

```xml
  ...
  <node pkg="nodelet" type="nodelet" name="foxglove_bridge"
        args="load foxglove_bridge/foxglove_bridge_nodelet $(arg nodelet_manager)">
    <rosparam>
      port: 8765
      topic_whitelist:
        - "^(?!/camera/image$).*"
    </rosparam>
  </node>
```

PR #158 (2023-02-13): Improve launch file documentation, add missing launch file arguments
**Public-Facing Changes**
- Add missing launch file arguments
- Improve launch file documentation

**Description**
Adds missing launch file arguments and adds some documentation on how to pass arguments to the launch file when launching via CLI or when included in another launch file.

Fixes #157 ([FG-1854](https://linear.app/foxglove/issue/FG-1854/docs-are-a-little-sparse-on-configuration-[foxgloveros-foxglove-bridge))
commit f0168ab7 (2023-02-13): Improve launch file documentation, add missing launch file arguments (#158)

**Public-Facing Changes**
- Add missing launch file arguments
- Improve launch file documentation

**Description**
Adds missing launch file arguments and adds some documentation on how to
pass arguments to the launch file when launching via CLI or when
included in another launch file.

Fixes #157
([FG-1854](https://linear.app/foxglove/issue/FG-1854/docs-are-a-little-sparse-on-configuration-[foxgloveros-foxglove-bridge))
closed 2023-02-13 by jhurliman