foxglove/ros-foxglove-bridge #164: [ROS2]

bertaveira, 2023-02-16:
**Description**
Right now, we have a custom service type that we do not really care about. The problem is that the bridge tries to get it and since it does not recognise the type it throws errors on the terminal as well as warnings on the ros log. While this happens we lose messages for about 3 seconds until it comes back to life.

Log window:
[image]

- Version: Latest main
- Platform: Ubuntu 22.04 (ROS humble)

bertaveira, 2023-02-16:
We tried adding the custom message but now the error is in retrieving the parameters and it still stops recieving for a few seconds 

[image]

achim-k, 2023-02-16:
There should not be a delay services / parameters could not be retrieved. I will look into this.

For the short term, you may want to whitelist only the services / parameters that you are interested in

PR #165 (2023-02-17): Run client handler functions in separate thread
**Public-Facing Changes**
- Run client handlers in separate thread

**Description**
Builds on top of #156 

So far we have executed client handler functions in the websocket server thread. This could cause issues with clients not receiving messages anymore while a longer-running handler function was executing. This PR fixes this by handler functions being executed in a separate thread.

Fixes #164 
Fixes FG-2029
commit 997adb99 (2023-02-17): Run client handler functions in separate thread (#165)

**Public-Facing Changes**
- Run client handlers in separate thread

**Description**
Builds on top of #156 

So far we have executed client handler functions in the websocket server
thread. This could cause issues with clients not receiving messages
anymore while a longer-running handler function was executing. This PR
fixes this by handler functions being executed in a separate thread.

Fixes #164 
Fixes FG-2029
closed 2023-02-17 by achim-k