foxglove/ros-foxglove-bridge #166: recvUC: malformed packet received from vendor 1.16

tonynajjar, 2023-02-17:
**Description**

- Version:
1. foxglove bridge: 0.3.0
2. foxglove: 1.40.0
- Platform: Ubuntu 22

**Steps To Reproduce**

1. have foxglove bridge running and connect foxglove to it
2. Create a Teleop panel publishing to `/test/

[image]

3. `ros2 topic echo /test` in a terminal:

```
root@logi-XMG-CORE-REN-M20:/code# ros2 topic echo /test
1676632965.771806 [0]     recvUC: malformed packet received from vendor 1.16 state parse:HEARTBEAT <52545053 02010110 e8b81001 c0bcc7f7 999987f0 09010800 8563ef63 7dfa8fc5 @0x74 07011c00 00000000 00006903 00000000 0e000000 00000000 0e000000 ad010000> (note: maybe partially bswap'd) smid 0x7 flags 0x1 otnh 28 rid 0x0 wid 0x3690000 first 14 last 14
....
```

**Expected Behavior**

no error and receive messages

achim-k, 2023-02-17:
I can reproduce this with the latest humble release and https://studio.foxglove.dev 

The problem is, that the latest release is quite outdated. It works with the latest source build which includes https://github.com/foxglove/ros-foxglove-bridge/pull/134. I believe the issue is similar to https://github.com/foxglove/studio/issues/5270#issuecomment-1430220382.

For now I would recommend building from source. We are about to release a new version, but it will take some time for the next syncs to happen.

achim-k, 2023-02-23:
Closing, as it works when building `foxglove_bridge` from source. If you do not want to build the package from source, you will have to wait until the latest `0.4.1` release is available from the official ROS packages

closed 2023-02-23 by achim-k