foxglove/ros-foxglove-bridge #171: [ROS2] Failed to parse channel schema

bertaveira, 2023-02-20:
Hi! 

We just updated to the newest version of the Ros Foxglove Bridge (commit 8f327d1) and are experiencing issues reading topics. It worked fine on the version we were using previously.

Most of the topics are common schemes such as Float32 and Float64 etc. See image below:
[image]

Pressing one of the errors gives the following output, this is a Float64 message:
[image]

achim-k, 2023-03-03:
@bertaveira is this still an issue for you?

achim-k, 2023-03-06:
Closing this for now. @bertaveira feel free to re-open if the problem persists.

closed 2023-03-06 by jtbandes