foxglove/ros-foxglove-bridge #174: Missing state transition messages

rgov, 2023-02-27:
```
ros-noetic-foxglove-bridge/now 0.3.0-1focal.20230106.232224 arm64 [installed,local]
```

Streaming data from a system to Foxglove Studio (1.39.1; Darwin), I was surprised that my state transition panel seemed to be stuck on the same state since connection, even though the system was progressing through other states.

When I pulled the rosbag afterwards, I saw indeed the state transition messages were being sent:

[image]

(I only broadcast the message once per state transition, I don't spam the current state to the topic.)

Other messages seemed to get through; I had an image panel that updated occasionally as expected (though I didn't watch it carefully).

What is the behavior regarding when messages get dropped? If I have a high bandwidth topic, will it elbow out small topics?

achim-k, 2023-02-27:
> What is the behavior regarding when messages get dropped? If I have a high bandwidth topic, will it elbow out small topics?

The [current behavior](https://github.com/foxglove/ros-foxglove-bridge/blob/f4b2f68cde3cc87ad65cb95a36cd7690da34b3a5/foxglove_bridge_base/include/foxglove_bridge/websocket_server.hpp#L1019-L1028) is rather simple and does not take into account which topic a message originates from. This is something we have to improve on in the future.

Are you sure that messages are dropped? Do you see the corresponding log messages?

On another note, the `0.3-0` release is rather outdated. I would recommend building the latest version from source or to wait for the next [noetic sync](https://discourse.ros.org/t/preparing-for-noetic-sync-2023-02-23/29929) (which seems to be just a couple of days away)

achim-k, 2023-03-09:
@rgov do you have any update on this? Do you still have this issue?

rgov, 2023-03-09:
Has anything changed with the algorithm? I haven't really used it since filing this. My system is offline for a few days.

achim-k, 2023-03-09:
> Has anything changed with the algorithm?

No it's still the same. But we are definitely open to improve on that. Have you had the same issue with `rosbridge`?

defunctzombie, 2023-03-23:
Since the ros-foxglove-bridge server side can get backed-up under certain workloads, I think a reasonable enhancement to resolve this poor UX experience would be for the server to send a warning [Status](https://github.com/foxglove/ws-protocol/blob/main/docs/spec.md#status) when it drops data. While this won't solve the problem of dropping the data, this will give visibility into what happened so you don't think "this panel has failed" but instead realize "oh my data connection could not keep up".