foxglove/ros-foxglove-bridge #180: [ROS1] Topic whitelist is not bidirectional

sseeberger96, 2023-03-06:
**Description**

We are evaluating using `foxglove_bridge` for forwarding data from a ROS1 stack that serves as a "gateway" for communicating with our robot to remote clients running Foxglove Studio. 

One feature that is especially important to us is the topic whitelist, since we want the ability to selectively control how much data gets sent from the gateway stack to the clients. However, we also want to limit the publishing capabilities from client to gateway. This is necessary because we have custom buttons in our Foxglove Studio layout that send commands to the robot (abort mission, change configuration, etc.) and we'd like to have a "read-only" mode for our UI that makes these buttons non-functional. We thought we could accomplish this via the topic whitelist by simply black-listing the topics that contain the control output from these button presses. However, in testing the bridge we found that the whitelist is not bi-directional, so publishing requests from client to the websocket server are un-guarded and make it through regardless of any specified topic blacklists. 

We made a patch for this issue that makes the whitelist bi-directional and are currently testing it (initial results indicate that it does work for blacklisting the publishes from our button presses). Is there a design reason that the whitelist is not currently configured to be bi-directional? If not, we'd be happy to open a PR to upstream the patch.

- Version: latest source
- Platform: ROS1 Melodic, Ubuntu 18.04, ARM device

jtbandes, 2023-03-06:
Hi, thanks for starting the discussion. I have a couple follow-up questions based on your writeup:
- Do you need to disable client publishing of specific topics, or just all topics? Would it be sufficient to provide a server configuration that completely disables client publishing?
- Since Foxglove Studio does not have any notion of read-only topics in the app, how do you plan to implement the "read-only mode" in your UI? Will your buttons appear disabled, or will they just do nothing when clicked?

sseeberger96, 2023-03-07:
Hi, thanks for the quick response! 

- For our current use case, I think a server configuration that disables all client publishing would be sufficient. Since we just want a means of remotely visualizing data in Foxglove Studio without giving the ability to command the robot to take action, I can't imagine us wanting the client to publish on any topics. For future use cases, though, I could see a potential for wanting the client to be able to publish on some, but not all, topics (e.g. gating allowed commands by user level).
- This is still up for discussion in our group, but for a basic first pass the thought is to have the buttons just do nothing when clicked, since that requires no code changes. We might decide to change that as we go forward.

achim-k, 2023-03-07:
> We made a patch for this issue that makes the whitelist bi-directional and are currently testing it (initial results indicate that it does work for blacklisting the publishes from our button [...]

I should have read the last sentence earlier, before implementing it myself. Anyway, @sseeberger96 , would be great if you could take a look at #181.

I will also submit a PR that allows to prevent clients from publishing to topics or calling services

achim-k, 2023-03-07:
Will keep this issue open until #181 got merged

closed 2023-03-08 by achim-k

sseeberger96, 2023-03-27:
Sorry for the delayed response here @achim-k, but we tested the latest updates on our system recently and the features added in #181 worked perfectly for us! Thanks so much for the support!

PR #182 (2023-03-07): Make server capabilities configurable
### Public-Facing Changes

- Make server capabilities configurable

### Description
Allows the user to configure server capabilities via a ROS parameter. By default, all capabilities are enabled. Note that the `time` capability is not configurable as it depends solely on `/use_sim_time`. 

Example usage:

```bash
roslaunch foxglove_bridge foxglove_bridge.launch capabilities:=[parameters,connectionGraph]
```

Fixes #180
commit de37eb67 (2023-03-07): Make server capabilities configurable (#182)

### Public-Facing Changes

- Make server capabilities configurable

### Description
Allows the user to configure server capabilities via a ROS parameter. By
default, all capabilities are enabled. Note that the `time` capability
is not configurable as it depends solely on `/use_sim_time`.

Example usage:

```bash
roslaunch foxglove_bridge foxglove_bridge.launch capabilities:=[parameters,connectionGraph]
```

Fixes #180
PR #181 (2023-03-08): Add parameter `client_topic_whitelist` for whitelisting client-published topics
### Public-Facing Changes
- Add parameter `client_topic_whitelist` for whitelisting client-published topics

### Description
Fixes #180
commit 6d418063 (2023-03-08): Add parameter `client_topic_whitelist` for whitelisting client-published topics (#181)

### Public-Facing Changes
- Add parameter `client_topic_whitelist` for whitelisting
client-published topics

### Description
Fixes #180