foxglove/ros-foxglove-bridge #200: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

sgf201, 2023-03-17:
Description
create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'
the topic is not advertised

    Version:ROS2 humble bridge with WebSocket++/0.8.2
    Platform: ubuntu22

Steps To Reproduce

Qos set to rclcpp::QoS qos(rclcpp::KeepLast{7});

Expected Behavior
topic advertised

            [rcutils|error_handling.c:108] rcutils_set_error_state()
            This error state is being overwritten:

'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.102152168] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[WARN] [1678937020.102373126] [foxglove_bridge]: Ignoring client unadvertisement from 192.168.6.195:52050 for unknown channel 3, client has 0 advertised topic(s)
[INFO] [1678937020.104918817] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 4
[INFO] [1678937020.105093695] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 4
[INFO] [1678937020.109083689] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 5
[INFO] [1678937020.110877709] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 6
[INFO] [1678937020.111063987] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 5
[INFO] [1678937020.111169025] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /clicked_point (0 subscribers) on channel 6

            [rcutils|error_handling.c:108] rcutils_set_error_state()
            This error state is being overwritten:

'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.114041753] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116

            [rcutils|error_handling.c:108] rcutils_set_error_state()
            This error state is being overwritten:

'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.116081249] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[INFO] [1678937020.118165425] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 9

I compiled the newest code  and test  result is
[foxglove_bridge-1] [INFO] [1679019306.793229602] [foxglove_bridge]: Client 192.168.6.195:54530 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 5
[foxglove_bridge-1] [INFO] [1679019306.793318962] [foxglove_bridge]: Client 192.168.6.195:54530 is no longer advertising /clicked_point (0 subscribers) on channel 6
[foxglove_bridge-1] 
[foxglove_bridge-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[foxglove_bridge-1] This error state is being overwritten:
[foxglove_bridge-1] 
[foxglove_bridge-1]   'Unknown QoS history policy, at ./src/qos.cpp:61'
[foxglove_bridge-1] 
[foxglove_bridge-1] with this new error message:
[foxglove_bridge-1] 
[foxglove_bridge-1]   'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'
[foxglove_bridge-1] 
[foxglove_bridge-1] rcutils_reset_error() should be called after error handling to avoid this.
[foxglove_bridge-1] <<<
[foxglove_bridge-1] [ERROR] [1679019306.796146855] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[foxglove_bridge-1] 
[foxglove_bridge-1] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[foxglove_bridge-1] This error state is being overwritten:
[foxglove_bridge-1] 
[foxglove_bridge-1]   'Unknown QoS history policy, at ./src/qos.cpp:61'
[foxglove_bridge-1] 
[foxglove_bridge-1] with this new error message:
[foxglove_bridge-1] 
[foxglove_bridge-1]   'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'
[foxglove_bridge-1] 
[foxglove_bridge-1] rcutils_reset_error() should be called after error handling to avoid this.
[foxglove_bridge-1] <<<
[foxglove_bridge-1] [ERROR] [1679019306.798299344] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[foxglove_bridge-1] [INFO] [1679019306.800405554] [foxglove_bridge]: Client 192.168.6.195:54530 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 9
[foxglove_bridge-1] [INFO] [1679019307.282059606] [foxglove_bridge]: Removed channel 3 for topic "/clicked_point" (geometry_msgs/msg/PointStamped)

achim-k, 2023-03-30:
> Steps To Reproduce
>
> Qos set to rclcpp::QoS qos(rclcpp::KeepLast{7});

I don't understand what you mean with that. Could you provide more detailed instructions on how to reproduce this issue?

defunctzombie, 2023-04-05:
@achim-k It looks to me like https://github.com/foxglove/ros-foxglove-bridge/pull/205 fixes this issue?

achim-k, 2023-04-05:
Yes, I forgot close this issue.

closed 2023-04-05 by achim-k

jayv, 2023-06-14:
@achim-k can we re-open, despite whatever was done in #205 I'm still seeing the exact same `create_publisher() failed setting topic QoS` messageon `v0.6.2` (0.6.2-1jammy.20230512.142701) which I believe #205 was merged into. 
The problem can be reproduced with something as simple as running the talker and listener demos together with foxglove web and bridge and publishing to `/chatter` from the Foxglove publisher panel. The problem doesn't happen when you don't spin up the talker, and have Foxglove be the first publisher for that topic. 
I'm using the  `osrf/ros:humble-desktop:latest` , `ghcr.io/foxglove/studio:latest` and bridge installed in another container based on humble-desktop.