foxglove/ros-foxglove-bridge #204: Publishers stay alive after client channel is unadvertised

achim-k, 2023-03-30:
**Description**

After unadvertising a client channel, the publisher is still registered in the ros2 graph

- Version: https://github.com/foxglove/ros-foxglove-bridge/commit/c4ccee324c7fabb53a8b15d50241ac911c459e1c
- Platform: ubuntu 22.04 / galactic & humble

**Steps To Reproduce**
1. Start foxglove server
2. Connect with a client  (e.g. `websocat`) and send the following commands
3. `{"op":"advertise", "channels":[{"id": 1, "topic": "achim", "encoding": "cdr", "schemaName": "std_msgs/msg/String"}]}`
4. confirm with `ros2 node info /foxglove_bridge` that the publisher has been created
5. `{"op":"unadvertise", "channelIds":[1]}`
6. `ros2 node info /foxglove_bridge` still shows the publisher

**Expected Behavior**
After the channel has been advertised, `ros2 node info /foxglove_bridge` should not show the publisher anymore

achim-k, 2023-03-30:
rclcpp issue: https://github.com/ros2/rclcpp/issues/2146

defunctzombie, 2023-04-21:
@achim-k The related linear issue is closed - does that mean this issue is done?

achim-k, 2023-04-21:
Yes, forgot to close

closed 2023-04-21 by achim-k