foxglove/ros-foxglove-bridge #208: Subscribers always using qos depth of 1 with fastDDS

eschembor-irobot, 2023-04-04:
**Description**
Subscribers created by the bridge always using qos depth of 1 with fastDDS

- Version: ca7d0c30910a97ccc56d00f25371d21e6da21eab
- Platform: Ubuntu 20.04

**Steps To Reproduce**
Currently, get_publishers_info_by_topic will report all publishers as having a qos depth of 0 with fastDDS. I opened a ticket for that here https://github.com/ros2/rmw_fastrtps/issues/684
This leads to foxglove_bridge using a qos depth of 1 for all subscribers when using fastDDS - which causes lots of dropped messages (a real headache for /tf visualization specifically). Hopefully, that issue with fastDDS will be resolved, but in the interim I suggest increasing the minimum qos depth for subscribers to alleviate this issue (internally, we're going to be patching foxglove_bridge for the time being to increase the minimum depth to 10).

**Expected Behavior**
Expect foxglove_bridge to use reasonable qos depth on subscribers when using fastDDS.

achim-k, 2023-04-06:
I guess a new parameter `min_qos_depth` (parameter `max_qos_depth` already exists) would be sufficient to solve this. Would that be an OK solution for you?

eschembor-irobot, 2023-04-06:
Yes, I think that's a good solution

PR #211 (2023-04-11): Add parameter for minimum subscription QoS depth
### Public-Facing Changes
- Add parameter for minimum subscription QoS depth [ROS2]

### Description
Adds an additional parameter for specifying the minimum QoS depth for subscriptions. Fixes #208
commit 6a18309d (2023-04-11): Add parameter for minimum subscription QoS depth (#211)

### Public-Facing Changes
- Add parameter for minimum subscription QoS depth [ROS2]

### Description
Adds an additional parameter for specifying the minimum QoS depth for
subscriptions. Fixes #208
closed 2023-04-11 by achim-k