foxglove/ros-foxglove-bridge #217: 'Unsupported parameter array type' breaks foxglove bridge

ysuurmeijer, 2023-04-20:
Receiving the following breaking error when trying to launch foxglove_bridge on a ROS network with a parameter of type array. The error results in shutdown of the nodelet manager. 

`[ERROR] [1682013405.757034088]: Invalid parameter: Parameter '/some_node/inputs': Unsupported parameter array type
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'`

Temporary workaround is to disable the parameterSubscribe capability `roslaunch foxglove_bridge foxglove_bridge.launch capabilities:=[clientPublish,parameters,services,connectionGraph]`

- Version: 0.5.3
- Platform: Linux / AMD64

Launch `foxglove-bridge` on a ROS network with and upload the following parameter file under the `some_node` namespace
```
inputs:
  - topic: /gps/fix
    message_type: NavSatFix
  - topic: /gps2/fix
    message_type: NavSatFix
  - topic: /gps3/fix
    message_type: NavSatFix
```

achim-k, 2023-04-20:
> The error results in shutdown of the nodelet manager.

Are you sure about that? It should just log an error message in that case

achim-k, 2023-04-20:
Crash has been fixed in #218 
Closing this issue in favor of #219

closed 2023-04-20 by achim-k