foxglove/ros-foxglove-bridge #220: [Foxglove Nodelet Manager] process had died

joewong00, 2023-04-28:
**Description**
Everything is working fine until I launch **robot localization** node package, then foxglove throws the error "terminate called after throwing an instance of 'XmlRpc::XmlRpcException'.

- Version:
- Platform: Jetson Xavier NX for robot, M1 macos for foxglove client

**Steps To Reproduce**

1. Launch foxglove bridge: `roslaunch foxglove_bridge foxglove_bridge.launch port:=8765`
2. Launch robot localization node with preset parameters: `roslaunch robot_localization robot_localization.launch`3. 
3. Establish connection from foxglove studio on client side

**Expected Behavior**
- Able to establish connection without any issue

defunctzombie, 2023-04-28:
Please provide the ROS version and the `foxglove_bridge` version.

joewong00, 2023-04-28:
ROS version: ROS melodic
Foxglove bridge version: 0.5.1

joewong00, 2023-04-28:
After some more digging into this issue, I realised that the error has nothing to do with the robot localization package itself, but instead the loading of the parameters, in my case the array parameters of process noise covariance and initial estimate covariance. How do I fix this?

defunctzombie, 2023-04-28:
This issue has already been fixed in the latest code: https://github.com/foxglove/ros-foxglove-bridge/pull/218.

ROS melodic is EOL and I don't think will get any more package updates. I recommend you build this package from source in your ROS workspace if you want to keep using melodic.

closed 2023-04-28 by defunctzombie