foxglove/ros-foxglove-bridge #236: report err when my node running

sgf201, 2023-06-25:
**Description**
[image]

- Version:
- Platform:

**Steps To Reproduce**

there are large msg transmit when err occur

**Expected Behavior**
do not report err

achim-k, 2023-06-26:
Does `ros2 param list `/camera_node` fail as well?

If you do not use Foxglove Studio's parameter panel, you can disable the [`parameters` and `parametersSubscribe`](https://github.com/foxglove/ros-foxglove-bridge/blob/29ac3989383dc9f2e1d6862c73738422a87d8736/ros2_foxglove_bridge/launch/foxglove_bridge_launch.xml#L16) capabilities. This will prevent these error messages from being displayed.

achim-k, 2023-07-03:
> Does `ros2 param list /camera_node` fail as well?

@sgf201 does this fail as well? Do you still have this issue?

sgf201, 2023-07-04:
the test board is not usable now, I will report after test

achim-k, 2023-07-17:
I'm going to close this issue. @sgf201 feel free to reopen this issue with detailed reproduction steps and the foxglove bridge version.

closed 2023-07-17 by achim-k