foxglove/ros-foxglove-bridge #253: Definition of ros2 schema x has changed during the server's runtime

KhalilSelyan, 2023-07-17:
**Description**
I'm using Autoware, and working on building a bunch  extensions and message converters for the 3D Panel as well,

[image]

- Version: 0.7.0
- Platform: Ubuntu 22.04 Jammy

**Steps To Reproduce**

**Expected Behavior**

achim-k, 2023-07-17:
It seems that foxglove bridge communicates a different schema for the builtin schemas from [@foxglove/rosmsg-msgs-common](https://github.com/foxglove/rosmsg-msgs-common). Is your foxglove bridge running correctly and able to find `.msg` files on your local disk? It would be helpful to get some logs or even better the websocket messages that the server sends to a new client when it connects. You can record the latter using [websocat](https://github.com/vi/websocat) with `websocat ws://localhost:8765 --protocol foxglove.websocket.v1 --no-close`

KhalilSelyan, 2023-07-24:
Hi yes, sorry for a late update on this i've been a bit busy.
this is a link to the output of the websocat command it is unformatted as i'm not sure exactly what to look for i apologize in advance, as well as a screenshot of what my launch file looks like after the last suggestion from the other issue 
https://gist.github.com/KhalilSelyan/04a67cf69f045d0be304ad643ac77329
PS: it hasn't been fixed it seems as it would still crash after just a period longer now than 15 seconds 

[image]

closed 2023-07-27 by achim-k

KhalilSelyan, 2023-07-27:
[image]
Quick update:
  I pulled latest main on studio, to test it out but sadly seems like hasn't quite done the trick.
  Some of those messages types errors went away like the Point and Quaternion and Vector3 and Point32, so i'm able to read some new messages but other message types still remain 😅

achim-k, 2023-07-27:
It would be the best if you could create an MCAP file that contains the schemas and some messages from your setup. The easiest way to do that is 
```
npx @foxglove/ws-protocol-examples@latest mcap-record ws://localhost:8765/ -o <file.mcap>
```

This connects, just like Studio, to foxglove_bridge and writes schema data + messages to the specifiied mcap file until you kill the process. Recording for a couple of seconds should be sufficient.

KhalilSelyan, 2023-07-27:
Here's a link to the mcap, little issue is some messages are idl and it just marked them as no schema it seems not sure if that's normal
https://drive.google.com/file/d/1qNaZfpUqJzjctNaxhaAFHdgS94OIgopG/view?usp=sharing

JonasHablitzel, 2023-09-12:
Hi,

is there a update on this topic? I'm facing the same issue with Foxglove **1.69.0**

achim-k, 2023-09-12:
> is there a update on this topic? I'm facing the same issue with Foxglove **1.69.0**

No, the issue has been resolved. What errors are you seeing? Would it be possible for you to record an mcap file that includes the schemas so I can reproduce this locally?

```
npx @foxglove/ws-protocol-examples@latest mcap-record ws://localhost:8765/ -o issue_253.mcap
```

JonasHablitzel, 2023-09-13:
here are the files:
[https://drive.google.com/file/d/12dst27fVaky-m2LcA0K57kj0Z8Dl-PhX/view?usp=drive_link](mcap)

i also created an example workspace wich publishes dummy messages and there it works for some reason, with the same message. the main problem is the autoware_auto_perception_msgs/msg/DetectedObjects type from the pointpillars implementation of Autoware.
[https://drive.google.com/file/d/19lIMM9aTTLs6wf4VMOLmRd4HE9LADsx2/view?usp=drive_link](issue_ws)
[image]

achim-k, 2023-09-13:
Regarding autoware, can you confirm that your version includes https://github.com/tier4/autoware_auto_msgs/pull/45? This was also the reason for the problems @KhalilSelyan encountered

JonasHablitzel, 2023-09-13:
Hello, thanks for the quick repsonse! A new pull in my other repo fixed the issue. so i can see now the messages