foxglove/ros-foxglove-bridge #254: Running the bridge with Autoware, makes it crash within 10-15seconds

KhalilSelyan, 2023-07-17:
**Description**
As you can see in this video here, as the bridge launches alongside Autoware after about 15seconds, it would just crash.
I've tried multiple times on different computers always same outcome of ros-foxglove-bridge crashing.

https://github.com/foxglove/ros-foxglove-bridge/assets/36904941/10fe7d9b-272e-4f58-9aae-84e9f9819282

- Version: 0.7.0
- Platform: Ubuntu 22.04 Jammy // ROS2 humble

**Steps To Reproduce**

Install autoware -- https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/
run the planning simulator -- https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/

run the ros-foxglove-bridge alongside it.

**Expected Behavior**
The bridge doesn't crash and keeps running

achim-k, 2023-07-17:
What middleware are you using? cyclone_dds? In that case the crash is probably due to https://github.com/ros2/rmw_cyclonedds/issues/268

Your node graph consists of many nodes and fetching parameters from all of these node spawns many threads. I'd recommend you to disable the [parameters](https://github.com/foxglove/ros-foxglove-bridge/blob/7fbd1ab1d86023c2a0a2cdbe98d78a1b0f16215e/ros2_foxglove_bridge/launch/foxglove_bridge_launch.xml#L16) capability if you do not require parameters to be inspectible in Studio.

KhalilSelyan, 2023-07-17:
That seems to actually have done the trick ! i'll keep testing for a while, but seems like so far it's not crashing. Cheers!

achim-k, 2023-07-17:
OK, good to hear. I'm going to close this issue then. Feel free to reopen if you still have problems.

closed 2023-07-17 by achim-k