foxglove/ros-foxglove-bridge #262: Failing to retrieve service type, brings node down

carlosatrios, 2023-09-01:
**Description**
The node aborts during repeated "Failed to retrieve service type" and crashes the node.

```
SUMMARY
========

PARAMETERS
 * /foxglove_bridge/address: 0.0.0.0
2023-09-01T21:43:49.718834923Z  * /foxglove_bridge/asset_uri_allowlist: ['package://(/?\\...
2023-09-01T21:43:49.718837287Z  * /foxglove_bridge/capabilities: ['clientPublish',...
 * /foxglove_bridge/certfile: 
 * /foxglove_bridge/client_topic_whitelist: ['.*']
2023-09-01T21:43:49.718846455Z  * /foxglove_bridge/keyfile: 
 * /foxglove_bridge/max_update_ms: 5000
 * /foxglove_bridge/param_whitelist: ['.*']
 * /foxglove_bridge/port: 8765
 * /foxglove_bridge/send_buffer_limit: 10000000
 * /foxglove_bridge/service_whitelist: ['.*']
 * /foxglove_bridge/tls: False
2023-09-01T21:43:49.718861954Z  * /foxglove_bridge/topic_whitelist: ['.*']
 * /foxglove_nodelet_manager/num_worker_threads: 0
 * /rosdistro: noetic
2023-09-01T21:43:49.718868607Z  * /rosversion: 1.16.0

NODES
  /
    foxglove_bridge (nodelet/nodelet)
    foxglove_nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://hood1-1-rios-roscore-service:80/
]2;/opt/rios/install/share/foxglove_bridge/launch/foxglove_bridge.launch http://hood1-1-rios-roscore-service:80/
process[foxglove_nodelet_manager-1]: started with pid [68]
process[foxglove_bridge-2]: started with pid [69]
2023-09-01T21:43:49.954738356Z [ERROR] [1693604629.954637736]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/get_loggers: std::future_error: Broken promise
[ERROR] [1693604629.960421416]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604629.963452575]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/reload_shaders: std::future_error: Broken promise
[ERROR] [1693604629.966846903]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/load_config: std::future_error: Broken promise
[ERROR] [1693604629.970424157]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/load_config_discarding_changes: std::future_error: Broken promise
[ERROR] [1693604629.974170433]: Failed to retrieve service type or service description of service /rviz_msiraider_455523_4051231953047898198/save_config: std::future_error: Broken promise
[ERROR] [1693604629.976027506]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_3355783/get_loggers: std::future_error: Broken promise
[ERROR] [1693604629.977745424]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_3355783/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604629.982635479]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_3360882/get_loggers: std::future_error: Broken promise
[ERROR] [1693604629.985879822]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_3360882/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604629.995735004]: Failed to retrieve service type or service description of service /tf_echo_1693005407704472780/get_loggers: std::future_error: Broken promise
[ERROR] [1693604629.997922693]: Failed to retrieve service type or service description of service /tf_echo_1693005407704472780/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604630.002444470]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/get_loggers: std::future_error: Broken promise
[ERROR] [1693604630.007448601]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/set_logger_level: std::future_error: Broken promise
2023-09-01T21:43:50.013426630Z [ERROR] [1693604630.013370403]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/reload_shaders: std::future_error: Broken promise
[ERROR] [1693604630.018472310]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/load_config: std::future_error: Broken promise
[ERROR] [1693604630.023164429]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/load_config_discarding_changes: std::future_error: Broken promise
[ERROR] [1693604630.028526369]: Failed to retrieve service type or service description of service /rviz_msiraider_475863_1805560526383919607/save_config: std::future_error: Broken promise
[ERROR] [1693604630.033903217]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_68845/get_loggers: std::future_error: Broken promise
[ERROR] [1693604630.038839359]: Failed to retrieve service type or service description of service /rqt_gui_cpp_node_68845/set_logger_level: std::future_error: Broken promise
[ WARN] [1693604630.291370231]: Failed to retrieve type of service /rqt_gui_cpp_node_11371/get_loggers
2023-09-01T21:43:50.543664253Z [ WARN] [1693604630.543572709]: Failed to retrieve type of service /rqt_gui_cpp_node_11371/set_logger_level
[ WARN] [1693604630.795988773]: Failed to retrieve type of service /rqt_gui_cpp_node_15179/get_loggers
[ WARN] [1693604631.049000466]: Failed to retrieve type of service /rqt_gui_cpp_node_15179/set_logger_level
[ WARN] [1693604631.302035504]: Failed to retrieve type of service /rqt_gui_cpp_node_15179/compressed/set_parameters
[ERROR] [1693604631.305001208]: Failed to retrieve service type or service description of service /tf_echo_1693329846940349580/get_loggers: std::future_error: Broken promise
[ERROR] [1693604631.307799466]: Failed to retrieve service type or service description of service /tf_echo_1693329846940349580/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604631.312068543]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/get_loggers: std::future_error: Broken promise
[ERROR] [1693604631.315257622]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/set_logger_level: std::future_error: Broken promise
2023-09-01T21:43:51.319103386Z [ERROR] [1693604631.318751839]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/reload_shaders: std::future_error: Broken promise
[ERROR] [1693604631.322952897]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/load_config: std::future_error: Broken promise
[ERROR] [1693604631.327608327]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/load_config_discarding_changes: std::future_error: Broken promise
[ERROR] [1693604631.331009648]: Failed to retrieve service type or service description of service /rviz_msiraider_613058_4222886344543863053/save_config: std::future_error: Broken promise
[ERROR] [1693604631.379363711]: Failed to retrieve service type or service description of service /rostopic_614368_1693332654733/get_loggers: std::future_error: Broken promise
[ERROR] [1693604631.383728559]: Failed to retrieve service type or service description of service /rostopic_614368_1693332654733/set_logger_level: std::future_error: Broken promise
[ERROR] [1693604631.386775559]: Failed to retrieve service type or service description of service /rviz_msiraider_618543_7783352244266021340/get_loggers: std::future_error: Broken promise
[... 131 lines cut ...]
2023-09-01T21:44:06.046421617Z [ERROR] [1693604646.046374779]: Failed to retrieve service type or service description of service /robot_2_dx_motion_interface/get_loggers: std::future_error: Broken promise
[ERROR] [1693604646.047603399]: Failed to retrieve service type or service description of service /robot_2_dx_motion_interface/set_logger_level: std::future_error: Broken promise
2023-09-01T21:44:06.048841758Z [ERROR] [1693604646.048795140]: Failed to retrieve service type or service description of service /robot_2_fanuc_rmi_client/get_loggers: std::future_error: Broken promise
[... cut ...]
2023-09-01T21:44:08.961000753Z [ INFO] [1693604629.834298210]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1693604629.856897093]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] is now available.
[ INFO] [1693604648.840659673]: Bond broken, exiting
``` 

- Version: 0.7.1
- Platform: Ubuntu 20:04 - ROS1 Noetic

**Steps To Reproduce**
Launch foxglove-bridge along side instances of RVIZ that are hosted on second machine. 

**Expected Behavior**
Exception is that the node continues on even if its unable to get specific services.

achim-k, 2023-09-04:
```
[ INFO] [1693604648.840659673]: Bond broken, exiting
```

It is the broken [bond](http://wiki.ros.org/bond) that causes the node to go down, without it `foxglove_bridge` would remain running (and logging all these errors).

Regardless, I have submitted #263 which I think should make service type retrieval a lot more robust. Could you give it a test drive?

Also, if you are not intending to use the `services` capability, you can simplly remove it from the `capabilities` list.

carlosatrios, 2023-09-05:
Thank you for the prompt response @achim-k! That does look to have resolved the issue with the node crashing.

carlosatrios, 2023-09-05:
Looks like its not fully fixed. We get another broken bond when we terminate a foxglove studio instance connected to this patched bridge. 

```
2023-09-05T16:49:55.569952054Z [ INFO] [1693932595.567760045]: [WS] Unsubscribing from parameter '/oak2/stereo_i_low_bandwidth'.
[ INFO] [1693932595.567765616]: [WS] Unsubscribing from parameter '/oak2/stereo_i_low_bandwidth_quality'.
[ INFO] [1693932595.567771116]: [WS] Unsubscribing from parameter '/oak2/stereo_i_spatial_filter_alpha'.
[ERROR] [1693932595.782057741]: Failed to retrieve service type or service description of service /rviz_msiraider_712938_3675954756021939125/reload_shaders: Failed to retrieve service type
[ INFO] [1693932595.567776657]: [WS] Unsubscribing from parameter '/oak2/stereo_i_spatial_filter[ INFO] [1693932555.211025287]: Loading nodelet /foxglove_bridge of type foxglove_bridge/foxglove_bridge_nodelet to manager foxglove_nodelet_manager with the following remappings:
2023-09-05T16:49:55.958292364Z [ INFO] [1693932555.230931624]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1693932555.253678450]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] is now available.
2023-09-05T16:49:55.958318914Z [ INFO] [1693932595.739912453]: Bond broken, exiting
2023-09-05T16:49:56.002073967Z [foxglove_bridge-2] process has finished cleanly
```

carlosatrios, 2023-09-05:
Trying with services disabled, we get:

```
master has started, initiating launch
... logging to /root/.ros/log/rios_1693934935/roslaunch-hood1-1-headnode-1-1.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
]2;/opt/rios/install/share/foxglove_bridge/launch/foxglove_bridge.launch
started roslaunch server http://hood1-1-headnode-1:40387/

SUMMARY
========
2023-09-05T17:58:56.277362044Z 
PARAMETERS
2023-09-05T17:58:56.277368045Z  * /foxglove_bridge/address: 0.0.0.0
 * /foxglove_bridge/asset_uri_allowlist: ['package://(/?\\...
 * /foxglove_bridge/capabilities: ['clientPublish',...
 * /foxglove_bridge/certfile: 
 * /foxglove_bridge/client_topic_whitelist: ['.*']
 * /foxglove_bridge/keyfile: 
2023-09-05T17:58:56.277387201Z  * /foxglove_bridge/max_update_ms: 5000
 * /foxglove_bridge/param_whitelist: ['.*']
 * /foxglove_bridge/port: 8765
2023-09-05T17:58:56.277396048Z  * /foxglove_bridge/send_buffer_limit: 10000000
 * /foxglove_bridge/service_whitelist: ['.*']
2023-09-05T17:58:56.277401980Z  * /foxglove_bridge/tls: False
 * /foxglove_bridge/topic_whitelist: ['.*']
2023-09-05T17:58:56.277407831Z  * /foxglove_nodelet_manager/num_worker_threads: 0
 * /rosdistro: noetic
2023-09-05T17:58:56.277413742Z  * /rosversion: 1.16.0

NODES
  /
2023-09-05T17:58:56.277451143Z     foxglove_bridge (nodelet/nodelet)
    foxglove_nodelet_manager (nodelet/nodelet)

ROS_MASTER_URI=http://hood1-1-rios-roscore-service:80/
2023-09-05T17:58:56.281547796Z ]2;/opt/rios/install/share/foxglove_bridge/launch/foxglove_bridge.launch http://hood1-1-rios-roscore-service:80/
2023-09-05T17:58:56.323674728Z process[foxglove_nodelet_manager-1]: started with pid [68]
process[foxglove_bridge-2]: started with pid [69]
[ WARN] [1693936736.417956380]: Could not find package 'control_msgs' in msg paths!
2023-09-05T17:58:56.418076828Z [ WARN] [1693936736.417983291]: Could not find definition for type control_msgs/FollowJointTrajectoryActionResult
[ WARN] [1693936736.418470826]: Could not find package 'control_msgs' in msg paths!
[ WARN] [1693936736.418482598]: Could not find definition for type control_msgs/FollowJointTrajectoryActionResult
2023-09-05T17:58:56.418651048Z [ WARN] [1693936736.418621492]: Could not find package 'control_msgs' in msg paths!
2023-09-05T17:58:56.418654064Z [ WARN] [1693936736.418631801]: Could not find definition for type control_msgs/FollowJointTrajectoryActionFeedback
2023-09-05T17:58:56.418686976Z [ WARN] [1693936736.418658913]: Could not find package 'control_msgs' in msg paths!
2023-09-05T17:58:56.418691395Z [ WARN] [1693936736.418667650]: Could not find definition for type control_msgs/FollowJointTrajectoryActionFeedback
2023-09-05T17:58:56.418897496Z [ WARN] [1693936736.418860716]: Could not find package 'dx_observer_msgs' in msg paths!
[ WARN] [1693936736.418870945]: Could not find definition for type dx_observer_msgs/Observation
[ERROR] [1693936746.521918680]: [WS] handle_accept error: Operation aborted.
[ INFO] [1693936736.357754332]: Initializing nodelet with 64 worker threads.
[ INFO] [1693936736.408834302]: Starting foxglove_bridge (noetic, 0.7.1@53ad89b) with WebSocket++/0.8.1
2023-09-05T17:59:06.527656057Z [ INFO] [1693936736.409003203]: [WS] Server running without TLS
[ INFO] [1693936736.409237186]: [WS] WebSocket server listening at ws://0.0.0.0:8765
2023-09-05T17:59:06.527674572Z [ INFO] [1693936736.409344370]: [WS] WebSocket server run loop started
[ INFO] [1693936739.164181210]: [WS] Client 100.123.190.98:40170 connected via /
2023-09-05T17:59:06.527687306Z [ INFO] [1693936739.517243286]: [WS] Subscribing to connection graph updates.
[ INFO] [1693936739.518012145]: Client 100.123.190.98:40170 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 1
[ INFO] [1693936739.518422734]: Client 100.123.190.98:40170 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 2
[ INFO] [1693936739.518816842]: Client 100.123.190.98:40170 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 3
[ INFO] [1693936739.826677673]: [WS] Client 10.100.0.106:57592 connected via /
2023-09-05T17:59:06.527735347Z [ INFO] [1693936740.160383853]: Client 10.100.0.106:57592 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 1
[ INFO] [1693936740.160431323]: Client 10.100.0.106:57592 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 2
2023-09-05T17:59:06.527748141Z [ INFO] [1693936740.160459857]: Client 10.100.0.106:57592 is no longer advertising /clicked_point (0 subscribers) on channel 1
[ INFO] [1693936740.160491778]: Client 10.100.0.106:57592 is no longer advertising /initialpose (0 subscribers) on channel 2
[ INFO] [1693936740.160518518]: Client 10.100.0.106:57592 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 3
[ INFO] [1693936740.160537134]: Client 10.100.0.106:57592 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 4
2023-09-05T17:59:06.527773119Z [ INFO] [1693936740.160559296]: Client 10.100.0.106:57592 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 5
[... 20 lines cut ...]
[ INFO] [1693936746.527620159]: Client 10.100.0.106:57592 is no longer advertising /clicked_point (0 subscribers) on channel 4
[... cut ...]
[ INFO] [1693936746.533980327]: Unsubscribing from topic "/tf_static" (tf2_msgs/TFMessage) on channel 347
2023-09-05T17:59:06.535732862Z [ INFO] [1693936746.534512356]: Unsubscribing from topic "/tf" (tf2_msgs/TFMessage) on channel 349

LOTS OF UNSUBSCRIBING...

[ INFO] [1693936736.353239575]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] has not been advertised, waiting...
2023-09-05T17:59:09.816259286Z [ INFO] [1693936736.374996878]: waitForService: Service [/foxglove_nodelet_manager/load_nodelet] is now available.
[ INFO] [1693936749.459915676]: Bond broken, exiting
```

achim-k, 2023-09-05:
It seems that the bond between the nodelet manager and the nodelet breaks, causing foxglove bridge to shutdown. You could try  changing the following line
https://github.com/foxglove/ros-foxglove-bridge/blob/ddff605b52cae3c0b3666768051b3d2933ce8600/ros1_foxglove_bridge/launch/foxglove_bridge.launch#L24
to 
```
args="load foxglove_bridge/foxglove_bridge_nodelet $(arg nodelet_manager) --no-bond"> 
```
to disable the bonding mechanism. 

Usually the bond should not break if the nodelet manager and nodelet loader are running on the same host (see also https://answers.ros.org/question/9700/nodelets-and-bond-timeouts/). If that's the case, try reducing the load by setting a `num_threads:=4` (should be enough) and disable all non-required capabilities

carlosatrios, 2023-09-05:
Disabling the bond now has the nodelet manager calling out its death:

```
[roslaunch][INFO] 2023-09-05 14:14:38,165: process[foxglove_bridge-2]: args[['/opt/ros/noetic/lib/nodelet/nodelet', 'load', 'foxglove_bridge/foxglove_bridge_nodelet', 'foxglove_nodelet_manager', '--no-bond', '__name:=foxglove_bridge']]
[roslaunch][INFO] 2023-09-05 14:14:38,165: ... created process [foxglove_bridge-2]
[roslaunch.pmon][INFO] 2023-09-05 14:14:38,165: ProcessMonitor.register[foxglove_bridge-2]
[roslaunch.pmon][INFO] 2023-09-05 14:14:38,165: ProcessMonitor.register[foxglove_bridge-2] complete
[roslaunch][INFO] 2023-09-05 14:14:38,165: ... registered process [foxglove_bridge-2]
[roslaunch][INFO] 2023-09-05 14:14:38,166: process[foxglove_bridge-2]: starting os process
[roslaunch][INFO] 2023-09-05 14:14:38,166: process[foxglove_bridge-2]: start w/ args [['/opt/ros/noetic/lib/nodelet/nodelet', 'load', 'foxglove_bridge/foxglove_bridge_nodelet', 'foxglove_nodelet_manager', '--no-bond', '__name:=foxglove_bridge', '__log:=/root/.ros/log/rios_1693934935/foxglove_bridge-2.log']]
[roslaunch][INFO] 2023-09-05 14:14:38,166: process[foxglove_bridge-2]: cwd will be [/root/.ros]
[roslaunch][INFO] 2023-09-05 14:14:38,169: process[foxglove_bridge-2]: started with pid [69]
[roslaunch][INFO] 2023-09-05 14:14:38,169: ... successfully launched [foxglove_bridge-2]
[roslaunch][INFO] 2023-09-05 14:14:38,169: ... launch_nodes complete
[roslaunch.pmon][INFO] 2023-09-05 14:14:38,169: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon 140441374443264)>
[roslaunch.parent][INFO] 2023-09-05 14:14:38,169: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2023-09-05 14:14:38,169: spin
[roslaunch][ERROR] 2023-09-05 14:14:59,908: [foxglove_nodelet_manager-1] process has died [pid 68, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=foxglove_nodelet_manager __log:=/root/.ros/log/rios_1693934935/foxglove_nodelet_manager-1.log].
```

carlosatrios, 2023-09-07:
No that is the full log, no additional context given. I think the bond breaking was pointing to the fact that the code is in fact dying. 

I can try that. 

Have you been able to reproduce on your end? f

carlosatrios, 2023-09-07:
@achim-k running as standalone, provided us with a few minutes of up-time, but it ultimately crashed with the same output. 

```
[foxglove_bridge-2] process has died [pid 69, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet standalone foxglove_bridge/foxglove_bridge_nodelet __name:=foxglove_bridge __log:=/root/.ros/log/rios_1693934935/foxglove_bridge-2.log].
log file: /root/.ros/log/rios_1693934935/foxglove_bridge-2*.log
```

achim-k, 2023-09-08:
> No that is the full log, no additional context given. I think the bond breaking was pointing to the fact that the code is in fact dying.

Could you launch with `roslaunch --screen`? This should show the output of the foxglove bridge node. Alternatively, add `output="screen"` in the launch file. Another alternative is, to launch the `foxglove_bridge` node (not the nodelet version) separately from the terminal:
```sh
rosrun foxglove_bridge foxglove_bridge  # _num_worker_threads:=4 _port:=8765
```

> Have you been able to reproduce on your end?

Unfortunately not :confused: Is there anything special about your setup (network setup, ros master on different host, ...)? 
It would be great to find out why the node is dying. Maybe you have some luck running foxglove bridge with gdb:
```bash
# Path might be different on your system
gdb -ex=run --args ../../install_ros1/lib/foxglove_bridge/foxglove_bridge _num_worker_threads:=4
```

carlosatrios, 2023-09-08:
GDB backtrace: 

```
Thread 7 "foxglove_bridge" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff19ed700 (LWP 980)]
0x00007ffff01cd7c6 in foxglove_bridge::retrieveServiceType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000l> >) () from /opt/rios/install/lib//libfoxglove_bridge_nodelet.so
(gdb) backtrace
#0  0x00007ffff01cd7c6 in foxglove_bridge::retrieveServiceType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000l> >) () from /opt/rios/install/lib//libfoxglove_bridge_nodelet.so
#1  0x00007ffff01a32f9 in foxglove_bridge::FoxgloveBridge::updateAdvertisedServices(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) ()
   from /opt/rios/install/lib//libfoxglove_bridge_nodelet.so
#2  0x00007ffff01ab9e8 in foxglove_bridge::FoxgloveBridge::updateAdvertisedTopicsAndServices(ros::TimerEvent const&) ()
   from /opt/rios/install/lib//libfoxglove_bridge_nodelet.so
#3  0x00007ffff7e85608 in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() () from /opt/ros/noetic/lib/libroscpp.so
#4  0x00007ffff7ea7172 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/noetic/lib/libroscpp.so
#5  0x00007ffff7ea7f15 in ros::CallbackQueue::callOne(ros::WallDuration) () from /opt/ros/noetic/lib/libroscpp.so
#6  0x00007ffff7fab2c9 in nodelet::detail::CallbackQueue::callOne() () from /opt/ros/noetic/lib/libnodeletlib.so
#7  0x00007ffff7fadd67 in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) ()
   from /opt/ros/noetic/lib/libnodeletlib.so
#8  0x00007ffff784043b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#9  0x00007ffff78eb609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#10 0x00007ffff7abb133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
``` 

Also here is the output of valgrind:
```
==1083== Thread 7:
==1083== Invalid read of size 8
==1083==    at 0xEAA87C6: foxglove_bridge::retrieveServiceType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000l> >) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0xEA7E2F8: foxglove_bridge::FoxgloveBridge::updateAdvertisedServices(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0xEA869E7: foxglove_bridge::FoxgloveBridge::updateAdvertisedTopicsAndServices(ros::TimerEvent const&) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0x49A6607: ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x49C8171: ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x49C8F14: ros::CallbackQueue::callOne(ros::WallDuration) (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x48A42C8: nodelet::detail::CallbackQueue::callOne() (in /opt/ros/noetic/lib/libnodeletlib.so)
==1083==    by 0x48A6D66: nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) (in /opt/ros/noetic/lib/libnodeletlib.so)
==1083==    by 0x4FC943A: ??? (in /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.71.0)
==1083==    by 0x4F15608: start_thread (pthread_create.c:477)
==1083==    by 0x4DA6132: clone (clone.S:95)
==1083==  Address 0x18 is not stack'd, malloc'd or (recently) free'd
==1083== 
==1083== 
==1083== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==1083==  Access not within mapped region at address 0x18
==1083==    at 0xEAA87C6: foxglove_bridge::retrieveServiceType(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::chrono::duration<long, std::ratio<1l, 1000l> >) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0xEA7E2F8: foxglove_bridge::FoxgloveBridge::updateAdvertisedServices(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0xEA869E7: foxglove_bridge::FoxgloveBridge::updateAdvertisedTopicsAndServices(ros::TimerEvent const&) (in /opt/rios/install/lib/libfoxglove_bridge_nodelet.so)
==1083==    by 0x49A6607: ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call() (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x49C8171: ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x49C8F14: ros::CallbackQueue::callOne(ros::WallDuration) (in /opt/ros/noetic/lib/libroscpp.so)
==1083==    by 0x48A42C8: nodelet::detail::CallbackQueue::callOne() (in /opt/ros/noetic/lib/libnodeletlib.so)
[... cut ...]
==1083==                       of which reachable via heuristic:
==1083==                         newarray           : 776 bytes in 1 blocks
==1083==         suppressed: 0 bytes in 0 blocks
==1083== Reachable blocks (those to which a pointer was found) are not shown.
==1083== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==1083== 
==1083== For lists of detected and suppressed errors, rerun with: -s
==1083== ERROR SUMMARY: 19 errors from 19 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```

achim-k, 2023-09-08:
Great, there is definitely something wrong. I'm suspecting that the connection ptr is a null pointer, and we currently do not check that one:
https://github.com/foxglove/ros-foxglove-bridge/blob/58b6712ab37486e435650927805d00bc7a78bbbf/ros1_foxglove_bridge/src/service_utils.cpp#L18

Would you be able to get another backtrace with debug symbols so we know exactly in which line number the segfault happens? You can do that by building with  `catkin build -DCMAKE_BUILD_TYPE=Debug`

carlosatrios, 2023-09-08:
backtrace with debug flags:

```
(gdb) backtrace
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff79be859 in __GI_abort () at abort.c:79
#2  0x00007ffff79be729 in __assert_fail_base (fmt=0x7ffff7b54588 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x7fffebfa3444 "px != 0", file=0x7fffebfa3418 "/usr/include/boost/smart_ptr/shared_ptr.hpp", line=734, 
    function=<optimized out>) at assert.c:92
#3  0x00007ffff79cffd6 in __GI___assert_fail (assertion=0x7fffebfa3444 "px != 0", file=0x7fffebfa3418 "/usr/include/boost/smart_ptr/shared_ptr.hpp", line=734, 
    function=0x7fffebfa3450 "typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = ros::ServiceServerLink; typename boost::detail::sp_member_access<T>::type = ros::ServiceServerLink*"...)
    at assert.c:101
#4  0x00007fffebf96f83 in boost::shared_ptr<ros::ServiceServerLink>::operator-> (this=0x7fffffffca50) at /usr/include/boost/smart_ptr/shared_ptr.hpp:734
#5  0x00007fffebf94803 in foxglove_bridge::retrieveServiceType (serviceName="/azure1/robot_state_publisher_azure/get_loggers", timeout=...) at /opt/rios/src/ros-foxglove-bridge/ros1_foxglove_bridge/src/service_utils.cpp:18
#6  0x00007fffebf1ba2e in foxglove_bridge::FoxgloveBridge::updateAdvertisedServices (this=0x5555558c5440, serviceNames=std::vector of length 297, capacity 512 = {...})
    at /opt/rios/src/ros-foxglove-bridge/ros1_foxglove_bridge/src/ros1_foxglove_bridge_nodelet.cpp:609
#7  0x00007fffebf19b9d in foxglove_bridge::FoxgloveBridge::updateAdvertisedTopicsAndServices (this=0x5555558c5440) at /opt/rios/src/ros-foxglove-bridge/ros1_foxglove_bridge/src/ros1_foxglove_bridge_nodelet.cpp:470
#8  0x00007fffebf13d59 in foxglove_bridge::FoxgloveBridge::onInit (this=0x5555558c5440) at /opt/rios/src/ros-foxglove-bridge/ros1_foxglove_bridge/src/ros1_foxglove_bridge_nodelet.cpp:184
#9  0x00007ffff7f864ad in nodelet::Nodelet::init(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, ros::CallbackQueueInterface*, ros::CallbackQueueInterface*) () from /opt/ros/noetic/lib/libnodeletlib.so
#10 0x00007ffff7f89a7b in nodelet::Loader::load(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&) () from /opt/ros/noetic/lib/libnodeletlib.so
#11 0x00005555555588fc in main (argc=1, argv=0x7fffffffd978) at /opt/rios/src/ros-foxglove-bridge/ros1_foxglove_bridge/src/ros1_foxglove_bridge_node.cpp:10
```

achim-k, 2023-09-08:
@carlosatrios could you give https://github.com/foxglove/ros-foxglove-bridge/pull/265 a try?

Judging from the backtrace, already [ros::ServiceManager::createServiceServerLink](https://github.com/ros/ros_comm/blob/030e132884d613e49a576d4339f0b8ec6f75d2d8/clients/roscpp/src/libros/service_manager.cpp#L235-L278) returns a invalid pointer. This can happen due to the following reasons:
1. [ROS is shutting down](https://github.com/ros/ros_comm/blob/030e132884d613e49a576d4339f0b8ec6f75d2d8/clients/roscpp/src/libros/service_manager.cpp#L241-L244)
2. [Lookup of the service server & port failed](https://github.com/ros/ros_comm/blob/030e132884d613e49a576d4339f0b8ec6f75d2d8/clients/roscpp/src/libros/service_manager.cpp#L248-L251)
3. [Connection to the service failed](https://github.com/ros/ros_comm/blob/030e132884d613e49a576d4339f0b8ec6f75d2d8/clients/roscpp/src/libros/service_manager.cpp#L275-L277)

In your case it's probably either 2. or 3. Maybe try enabling debug logging and see if you see any roscpp related error message.

closed 2023-09-11 by achim-k

carlosatrios, 2023-09-11:
Your fix looks to have worked. Thank you for all the help!

PR #263 (2023-09-05): Make ROS1 service type retrieval more robust
### Public-Facing Changes

Make ROS1 service type retrieval more robust

### Description
For ROS1, foxglove bridge has to retrieve the service type from the service server (by opening a connection) as the ROS master does not store the service type.

This patch makes service type retrieval more robust by
- Fixing the service link object getting out of scope too early
- Better exception handling
- Allowing users to set a custom timeout for service type retrieval

Fixes #262
commit 58b6712a (2023-09-05): Make ROS1 service type retrieval more robust (#263)

### Public-Facing Changes

Make ROS1 service type retrieval more robust

### Description
For ROS1, foxglove bridge has to retrieve the service type from the
service server (by opening a connection) as the ROS master does not
store the service type.

This patch makes service type retrieval more robust by
- Fixing the service link object getting out of scope too early
- Better exception handling
- Allowing users to set a custom timeout for service type retrieval

Fixes #262
PR #265 (2023-09-11): Fix invalid pointers not being caught
### Public-Facing Changes

Fix invalid pointers not being caught

### Description
- Fixes invalid pointers not being caught, leading to a segmentation fault as can be seen in https://github.com/foxglove/ros-foxglove-bridge/issues/262#issuecomment-1712148653
- Also exposes the `service_type_retrieval_timeout_ms` ROS1 parameter through the launch file and adds it to the readme (forgot that in #263)

Fixes #262 (hopefully)
commit b4f0a42f (2023-09-11): Fix invalid pointers not being caught (#265)

### Public-Facing Changes

Fix invalid pointers not being caught

### Description
- Fixes invalid pointers not being caught, leading to a segmentation
fault as can be seen in
https://github.com/foxglove/ros-foxglove-bridge/issues/262#issuecomment-1712148653
- Also exposes the `service_type_retrieval_timeout_ms` ROS1 parameter
through the launch file and adds it to the readme (forgot that in #263)

Fixes #262 (hopefully)