foxglove/ros-foxglove-bridge #267: crash in foxglove ros1 due to pointer

weidezhang, 2023-10-20:
**Description**
By restarting some of our ros nodes , we found foxglove_bridge_ros1 crash randomly. 

- Version:  foxglove_bridge_ros1 built from source from master branch (after 0.7.2) ros-noetic 
- Platform: Ubuntu 22.04

**Steps To Reproduce**
By bringing up and bringing down our ros nodes (with / without rviz), we found the following core dump : 
[New Thread 0x7ffff4202700 (LWP 4355)]
[New Thread 0x7ffff3a01700 (LWP 4356)]
[New Thread 0x7ffff3200700 (LWP 4357)]
[New Thread 0x7ffff29ff700 (LWP 4358)]
[New Thread 0x7ffff21fe700 (LWP 4359)]
[New Thread 0x7ffff19fd700 (LWP 4360)]
[New Thread 0x7ffff11fc700 (LWP 4361)]
[New Thread 0x7ffff09fb700 (LWP 4362)]
[New Thread 0x7fffe3fff700 (LWP 4363)]
[ INFO] [1697840735.025183026]: Initializing nodelet with 4 worker threads.
[New Thread 0x7fffe37fe700 (LWP 4364)]
[ INFO] [1697840735.344176016]: Starting foxglove_bridge (noetic, @) with WebSocket++/0.8.1
[New Thread 0x7fffe1606700 (LWP 4365)]
[ INFO] [1697840735.344597853]: [WS] Server running without TLS
[New Thread 0x7fffe0e05700 (LWP 4366)]
[New Thread 0x7fffdbfff700 (LWP 4367)]
[ INFO] [1697840735.344809370]: [WS] WebSocket server run loop started
[ INFO] [1697840735.344926661]: [WS] WebSocket server listening at ws://0.0.0.0:8765
[New Thread 0x7fffdb7fe700 (LWP 4368)]
[New Thread 0x7fffdaffd700 (LWP 4369)]
^B2[ INFO] [1697840740.832140313]: [WS] Client 127.0.0.1:35626 connected via /
[ INFO] [1697840740.834324626]: [WS] Subscribing to connection graph updates.
[ INFO] [1697840754.072684455]: Subscribed to topic "/image/color" (sensor_msgs/Image) on channel 20
[ INFO] [1697840754.073895438]: Subscribed to topic "/wcs_task_results" (chimp_communication_api/TaskResult) on channel 49
[ INFO] [1697840759.209122885]: Subscribed to topic "/cycle_status" (chimp_communication_api/CycleResult) on channel 62
[ INFO] [1697840759.210227291]: Subscribed to topic "/scanner_matched/color" (sensor_msgs/Image) on channel 61
[ INFO] [1697840769.277877571]: Subscribed to topic "/image/grasp_point_detection" (sensor_msgs/Image) on channel 67
[ INFO] [1697840769.278942044]: Subscribed to topic "/image/bin_localization" (sensor_msgs/Image) on channel 68
[New Thread 0x7fffda7fc700 (LWP 5085)]
[ WARN] [1697840784.279217726]: [WS] 127.0.0.1:35626: Client subscription id 5 did not exist; ignoring unsubscription
[ WARN] [1697840784.279259584]: [WS] 127.0.0.1:35626: Client subscription id 4 did not exist; ignoring unsubscription
[ERROR] [1697840789.660997503]: Failed to retrieve service type or service description of service /chimp_perception_node/set_logger_level: Timed out when retrieving service type

Thread 2 "foxglove_bridge" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff4202700 (LWP 4355)]
0x00007ffff01dd959 in std::__future_base::_Result_base::_Deleter::operator() (
    this=0x7ffff4201770, __fr=
    0x7fffe2f3b26b <foxglove::Server<foxglove::WebSocketNoTls>::updateConnectionGraph(std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocat--Type <RET> for more, q to quit, c to continue without paging--2
or<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > > const&)+6107>) at /usr/include/c++/9/future:213
213             void operator()(_Result_base* __fr) const { __fr->_M_destroy(); }
(gdb) bt
#0  0x00007ffff01dd959 in std::__future_base::_Result_base::_Deleter::operator() (
    this=0x7ffff4201770,
    __fr=0x7fffe2f3b26b <foxglove::Server<foxglove::WebSocketNoTls>::updateConnectionGraph(std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::unordered_set<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >  [... cut ...]
   from /opt/ros/noetic/lib/libroscpp.so
#13 0x00007ffff7efff67 in ros::PollSet::update(int) () from /opt/ros/noetic/lib/libroscpp.so
#14 0x00007ffff7e91dbd in ros::PollManager::threadFunc() ()
   from /opt/ros/noetic/lib/libroscpp.so
#15 0x00007ffff783543b in ?? () from /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0
#16 0x00007ffff78e0609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#17 0x00007ffff7ab0133 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95`

`

**Expected Behavior**
should not crash

achim-k, 2023-10-23:
@weidezhang I have submitted a potential fix in #268. Could you confirm if that fixes the crash?

weidezhang, 2023-10-23:
cool. seems the issue is fixed. i have verified locally. thx

PR #268 (2023-10-23): Fix callback accessing invalid reference to promise
### Public-Facing Changes

Fix callback accessing invalid reference to promise

### Description
For ROS1, the service request and response types can only be retrieved by opening a connection to the service server. For this, we use `ros::ServiceManager::instance()->createServiceServerLink` which under the hood keeps a copy of the returned pointer, effectively keeping the connection alive. This patch makes sure that we drop the connection when we either have received the header or when the timeout has been exceeded. This avoids that the header- [... cut]
commit e52bb9dd (2023-10-23): Fix callback accessing invalid reference to promise (#268)

### Public-Facing Changes

Fix callback accessing invalid reference to promise

### Description
For ROS1, the service request and response types can only be retrieved
by opening a connection to the service server. For this, we use
`ros::ServiceManager::instance()->createServiceServerLink` which under
the hood keeps a copy of the returned pointer, effectively keeping the
connection alive. This patch makes sure that we drop the connection when
we either have received the header or when the timeout has been
exceeded. This avoids that the [... cut]
closed 2023-10-23 by achim-k