foxglove/ros-foxglove-bridge #280: subscriber crashed when deserialize msg which is send by foxglove_bridge

sgf201, 2023-12-18:
**Description**
push button, in studio remote control panel, schema is Twist.subscriber crashed
 Stack trace (most recent call last):                                                                                                                                                           
[decodec-1] #12   Object "/home/admin/5v5r/fsinstall/nov_codec/lib/nov_codec/decodec", at 0xaaaae95184e3, in                                                                                               
[decodec-1] #11   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffff823bfe0f, in __libc_start_main                                                                                                       
[decodec-1] #10   Object "/home/admin/5v5r/fsinstall/nov_codec/lib/nov_codec/decodec", at 0xaaaae9518b57, in main                                                                                          
[decodec-1] #9    Object "/opt/nros/lib/librclcpp.so", at 0xffff828fad83, in rclcpp::executors::SingleThreadedExecutor::spin()                                                                             
[decodec-1] #8    Object "/opt/nros/lib/librclcpp.so", at 0xffff828f42fb, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)                                                              
[decodec-1] #7    Object "/opt/nros/lib/librclcpp.so", at 0xffff828f2fcb, in rclcpp::Executor::execute_subscription(std::shared_ptr<rclcpp::SubscriptionBase>)                                             
[decodec-1] #6    Object "/opt/nros/lib/librclcpp.so", at 0xffff828f208b, in                                                                                                                               
[decodec-1] #5    Object "/opt/nros/lib/librclcpp.so", at 0xffff828f1ee3, in                                                                                                                               
[decodec-1] #4    Object "/opt/nros/lib/librcl.so", at 0xffff8233183b, in rcl_take_loaned_message                                                                                                          
[decodec-1] #3    Object "/opt/nros/lib/librmw_cyclonedds_cpp.so", at 0xffff81f7bcd3, in                                                                                                                   
[decodec-1] #2    Object "/opt/nros/lib/librmw_cyclonedds_cpp.so", at 0xffff81f7b5a7, in rmw_deserialize
[decodec-1] #1    Object "/opt/nros/lib/librmw_cyclonedds_cpp.so", at 0xffff81f88b07, in 
[decodec-1] #0    Object "/opt/nros/lib/librmw_cyclonedds_cpp.so", at 0xffff81f88a84, in 
[decodec-1] Segmentation fault (Address not mapped to object [(nil)])
- Version:
latest
- Platform:
aarch64 
**Steps To Reproduce**

1.change rmw to cyclonedds + iceoryx and use loaned api

**Expected Behavior**
subscriber got msg,and work correctly

sgf201, 2023-12-18:
> > change rmw to cyclonedds + iceoryx and use loaned api将 RMW 更改为 CyclonedDS + IceoryX 并使用借出的 API
> 
> I have the felling that it is related to the RMW. Does it work when you use the default RMW?我听说它与RMW有关。当您使用默认 RMW 时，它是否有效？

it worked if I turn off loaned api,but I have to use loaned api for big msg

tonynajjar, 2024-02-09:
btw @sgf201 your issue is referencing many other issues because it's not properly formatting. You might want to put it in a code block

achim-k, 2024-03-22:
@sgf201 is this still an issue for you?

sgf201, 2024-03-24:
are the button msg senders of bridge using rclcpp to send msg？

achim-k, 2024-03-25:
Yes, we use `rclcpp` for that. The relevant code is 

https://github.com/foxglove/ros-foxglove-bridge/blob/e37a3e1e309c1a6255e3df5a86324e002dab65e1/ros2_foxglove_bridge/src/ros2_foxglove_bridge.cpp#L628

and

https://github.com/foxglove/ros-foxglove-bridge/blob/e37a3e1e309c1a6255e3df5a86324e002dab65e1/ros2_foxglove_bridge/src/ros2_foxglove_bridge.cpp#L711-L712

Can you spot any issues with that?

achim-k, 2024-04-05:
@sgf201 I'm going to close this issue since this seems to be related to a middleware issue.

> change rmw to cyclonedds + iceoryx and use loaned api

Judging from this, it also seems like you have to make changes to foxglove bridge to use the loaned API which it does not by default.

closed 2024-04-05 by achim-k

sgf201, 2024-06-13:
The root cause has been identified. When the bridge is using the generic publisher to forward messages, it does not use the publish_as_loaned interface for the original loan-based messages.
`  explicit PlayerPublisher(
      std::shared_ptr<rclcpp::GenericPublisher> pub,
      bool disable_loan_message)
    : publisher_(std::move(pub))
    {
      using std::placeholders::_1;
      if (disable_loan_message || !publisher_->can_loan_messages()) {
        publish_func_ = std::bind(&rclcpp::GenericPublisher::publish, publisher_, _1);
      } else {
        publish_func_ = std::bind(&rclcpp::GenericPublisher::publish_as_loaned_msg, publisher_, _1);
      }`

achim-k, 2024-07-01:
@sgf201 could you give #314 a try and check if it works?

PR #314 (2024-07-03): Support publishing client message as loaned message
### Changelog
Support publishing client message as loaned message

### Docs
None

### Description
Support publishing client message as loaned message when the middleware supports it. 
[`publish_as_loaned_msg`](https://docs.ros.org/en/humble/p/rclcpp/generated/classrclcpp_1_1GenericPublisher.html#_CPPv4N6rclcpp16GenericPublisher21publish_as_loaned_msgERKN6rclcpp17SerializedMessageE) is available as of ROS humble.

Fixes #280
commit dbb033e0 (2024-07-03): Support publishing client message as loaned message (#314)

### Changelog
Support publishing client message as loaned message

### Docs
None

### Description
Support publishing client message as loaned message when the middleware
supports it.

[`publish_as_loaned_msg`](https://docs.ros.org/en/humble/p/rclcpp/generated/classrclcpp_1_1GenericPublisher.html#_CPPv4N6rclcpp16GenericPublisher21publish_as_loaned_msgERKN6rclcpp17SerializedMessageE)
is available as of ROS humble.

Fixes #280