foxglove/ros-foxglove-bridge #287: Switching from one 3D tab to another does not subscribe to /tf

tonynajjar, 2024-02-20:
**Description**

Version:
- foxglove studio: 1.87.0
- foxglove bridge 0.7.4

**Steps To Reproduce**

1- In foxglove studio open 3 Tabs: 2 with 3D panel visualizing a TF tree and 1 unrelated (e.g. diagnostics)
[image]

2- Select one of the 3D tabs
3- Select the other 3D tab
4- See that not all not the Transforms are available

However when switching from the unrelated tab to the 3D tabs, then all the transforms are there.

Here is the log when switching from unrelated tab to 3D tab:

```
simulation  | [foxglove_bridge-33] [INFO] [1708422174.552680920] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 279
simulation  | [foxglove_bridge-33] [INFO] [1708422174.555097823] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 280
simulation  | [foxglove_bridge-33] [INFO] [1708422174.557606155] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 281
simulation  | [foxglove_bridge-33] [INFO] [1708422174.725017269] [foxglove_bridge]: Subscribing to topic "/tf" (tf2_msgs/msg/TFMessage) on channel 1
simulation  | [foxglove_bridge-33] [INFO] [1708422174.728160688] [foxglove_bridge]: Subscribing to topic "/tf_static" (tf2_msgs/msg/TFMessage) on channel 30
```

Here is the log when switching from a 3D tab to the other:

```
simulation  | [foxglove_bridge-33] [INFO] [1708422241.137217508] [foxglove_bridge]: Client 127.0.0.1:52808 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 279
simulation  | [foxglove_bridge-33] [INFO] [1708422241.137287078] [foxglove_bridge]: Client 127.0.0.1:52808 is no longer advertising /clicked_point (0 subscribers) on channel 280
simulation  | [foxglove_bridge-33] [INFO] [1708422241.137299441] [foxglove_bridge]: Client 127.0.0.1:52808 is no longer advertising /initialpose (0 subscribers) on channel 281
simulation  | [foxglove_bridge-33] [INFO] [1708422241.147544581] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 282
simulation  | [foxglove_bridge-33] [INFO] [1708422241.150091867] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 283
simulation  | [foxglove_bridge-33] [INFO] [1708422241.153672703] [foxglove_bridge]: Client 127.0.0.1:52808 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 284
```

You can see that there is no subscription to /tf and /tf_static in this case.

Additionally here is my foxglove bridge config:

```
    launch_foxglove_bridge = IncludeLaunchDescription(
        XMLLaunchDescriptionSource(
            join(share_foxglove_bridge, "launch/foxglove_bridge_launch.xml"),
        ),
        launch_arguments={
            # action server inbuilt messages are currently not supported
            # https://github.com/foxglove/ros-foxglove-bridge/issues/11
            "topic_whitelist": '["^(?:(?!_action).)*$"]',
            "service_whitelist": '["^(?:(?!_action).)*$"]',
            "capabilities": "[clientPublish]",
            "use_sim_time": conf_sim_time,
            "min_qos_depth": "10",
            "max_qos_depth": "100"
        }.items(),
    )
```

**Expected Behavior**

Transforms always subscribed to.

tonynajjar, 2024-03-19:
Hey, any initial investigation?

achim-k, 2024-03-19:
Sorry, this fell of the radar. I will look into this next week

achim-k, 2024-03-28:
It turns out, that when switching from one 3D-panel tab to another tab with a 3D panel, Foxglove Studio does not re-subscribe to `/tf_static` as it is already subscribed to it. However, since `/tf_static` is latched, the new panel does not receive these latched transforms. The reason it works when switching to a non-3D panel tab inbetween is because Studio unsubscribes and re-subscribes to `/tf_static. 

This is something we have to fix on the Studio side. However, I will keep this issue open until we have it fixed.

tonynajjar, 2024-07-12:
Hi @achim-k, friendly ping for an update please :slightly_smiling_face:

achim-k, 2024-07-12:
Internal PRs are in review, I hope that it makes it into the next release but cannot promise anything

achim-k, 2024-08-05:
@tonynajjar this should be working now in the web app. It would be great if you could give it a test and report back if there are any  remaining issues.

closed 2024-08-05 by achim-k

tonynajjar, 2024-08-06:
ok thanks, I tried it quickly, looks like it works!