foxglove/ros-foxglove-bridge #301: Invalid service definition

tonynajjar, 2024-06-11:
**Description**

- Version of foxglove bridge: 0.7.7
- Platform: foxglove studio 2.6.0

**Steps To Reproduce**

```
[foxglove_bridge-22] [WARN] [1718109515.363661001] [foxglove_bridge]: Failed to add service /perception/front_left/device_info;(realsense2_camera_msgs/srv/DeviceInfo): Invalid service definition in DeviceInfo.srv: Expected 2 definitions, got 1
```

The message seems reasonable https://github.com/IntelRealSense/realsense-ros/blob/ros2-development/realsense2_camera_msgs/srv/DeviceInfo.srv

achim-k, 2024-06-11:
Yeah, that's strange. I will take a look, thanks for reporting

tonynajjar, 2024-06-11:
I'm on Humble. I'm reverting your commit and checking if it reproduces

tonynajjar, 2024-06-11:
confirmed, checked out the commit before the PR I linked (hash 3e35af32fde80dc42e6c7bea40a2cc32c30170f4) and the issue does not reproduce

achim-k, 2024-06-11:
The reason is an additional CR `0x0D` before the actual line feed `0x0A` at the end of the line :sob: Will push a fix in a couple of hours.

[image]

tonynajjar, 2024-06-11:
It's impressive how fast you figured out such a hidden issue :laughing: Let me know when you have the fix, I might be still in time to give it a quick test

PR #303 (2024-06-11): Fix srv definition parsing failing due to carriage return
### Changelog
Fix srv definition parsing failing due to carriage return

### Docs
None

### Description
Fixes srv (or action) definition parsing failing when the definition seperator `---` is terminated with `\r\n` instead of just `\n`. More details in #301.

Fixes: #301
commit 4fcf6139 (2024-06-11): Fix srv definition parsing failing due to carriage return (#303)

### Changelog
Fix srv definition parsing failing due to carriage return

### Docs
None

### Description
Fixes srv (or action) definition parsing failing when the definition
seperator `---` is terminated with `\r\n` instead of just `\n`. More
details in #301.

Fixes: #301
closed 2024-06-11 by jtbandes