foxglove/ros-foxglove-bridge #308: websocket_server build gives `extra ‘;’ [-Werror=pedantic]`

karjavo, 2024-06-19:
**Description**

when compiling foxglove bridge for humble I encounter an error related to wpedantic and the presence of extra ;

```
--- stderr: foxglove_bridge
CMake Warning:
 Manually-specified variables were not used by the project:
 
CATKIN_INSTALL_INTO_PREFIX_ROOT

In file included from /opt/ros/humble/src/foxglove_bridge/foxglove_bridge_base/src/server_factory.cpp:5:
 /opt/ros/humble/src/foxglove_bridge/foxglove_bridge_base/include/foxglove_bridge/websocket_server.hpp:1047:2: error: extra ‘;’ [-Werror=pedantic]
  1047 | };
           |  ^
 cc1plus: all warnings being treated as errors
 make[2]: *** [CMakeFiles/foxglove_bridge_base.dir/build.make:132: CMakeFiles/foxglove_bridge_base.dir/foxglove_bridge_base/src/server_factory.cpp.o] Error 1
 make[1]: *** [CMakeFiles/Makefile2:117: CMakeFiles/foxglove_bridge_base.dir/all] Error 2
 make: *** [Makefile:146: all] Error 2

```
- Version: 0.7.7-1
- Platform: aarch64

**Steps To Reproduce**

juste trying install foxglove_bridge with rosdep :

```
rosinstall_generator --deps --rosdistro humble foxglove_bridge  > ros2.humble.rosinstall
cat ros2.humble.rosinstall
vcs import src < ros2.humble.rosinstall

rosdep update
rosdep install -y --ignore-src --from-paths src --rosdistro humble

colcon build --merge-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

**Expected Behavior**

jtbandes, 2024-06-20:
Thanks for the report, could you submit a PR to fix this?

apmilko, 2024-06-27:
> Thanks for the report, could you submit a PR to fix this?

Hi, I have created a PR to fix this issue. Can you please approve it? Don't know why, but some test for ROS rolling haven't passed, but I think that problem not in extra ";".

achim-k, 2024-06-27:
Closed by #311

AndBondStyle, 2024-07-05:
Can you push a new tag please? Installing via rosdep is still failing. Or is there a workaround maybe?

achim-k, 2024-07-05:
Why does `rosdep` fail? This PR should not be related to that.

AndBondStyle, 2024-07-05:
I think ROS only updates its package index when new tag is pushed (if I understand correctly). So it still gets the source code for v0.7.7-1 and fails
https://github.com/ros2-gbp/foxglove_bridge-release

achim-k, 2024-07-05:
Yes, version 0.7.8 hasn't been released yet. Why is that a problem for you? Are you depending on exactly that version or why is `rosdep` failing? Some log messages would be helpful