foxglove/ros-foxglove-bridge #310: foxglove crashes when loading large map (pgm files)

shannonwarren, 2024-06-26:
**Description**
Foxglove studio crashes when i load large maps through a ros2 service with the following error:

`Renderer process gone: { reason: 'crashed', exitCode: 134 }`

> shannon@shannon-XPS-15-9530:/usr/bin$ ./foxglove-studio 
initializing Sentry in main
Loaded language from settings: undefined
Set language: en
Foxglove Studio 2.1.0
Could not set app as handler for foxglove://
Elapsed (ms) until new StudioWindow: 65
New StudioWindow 1
window.loadURL(file:///opt/Foxglove Studio/resources/app.asar/renderer/index.html)
load-pending-files
injectFilesToOpen: no files - skipping
window URL loaded
Renderer process gone: { reason: 'crashed', exitCode: 134 }
window.loadURL(file:///opt/Foxglove Studio/resources/app.asar/renderer-crash-recovery/index.html)
load-pending-files
injectFilesToOpen: no files - skipping
window URL loaded
window.loadURL(file:///opt/Foxglove Studio/resources/app.asar/renderer/index.html)
load-pending-files
injectFilesToOpen: no files - skipping
window URL loaded

- Version: 2.1.0 
- Platform: Ubuntu 22.04

**Steps To Reproduce**

**Expected Behavior**

defunctzombie, 2024-06-26:
Please provide more specific steps to reproduce and an example file that was considered "large". In order to tackle this kind of bug report we need better details about the exact workflow and assets involved.

achim-k, 2024-07-15:
ping @shannonwarren

defunctzombie, 2024-07-26:
Closing for lack of response. @shannonwarren we are happy to re-visit this but need more steps and data.

closed 2024-07-26 by defunctzombie