foxglove/ros-foxglove-bridge #318: Default port and address of 'foxglove_bridge_launch.xml' not met when launching

qcharmoillau, 2024-07-24:
**Description**

- Version: ROS 2 Humble
- Platform: Raspberry Pi 5 (using Ubuntu 22.04) connected to a PC (using Windows)

**Steps To Reproduce**
1°) On the one hand, by using the command line `ros2 launch foxglove_bridge foxglove_bridge_launch.xml` in the Ubuntu terminal, the address and the port are properly set judging by the terminal's feedback :
```
[INFO] [foxglove_bridge-1]: process started with pid [13913]
[foxglove_bridge-1] [INFO] [1721823866.109491784] [foxglove_bridge_component_manager]: Load Library: /opt/ros/humble/lib/libfoxglove_bridge_component.so
[foxglove_bridge-1] [INFO] [1721823866.120097007] [foxglove_bridge_component_manager]: Found class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-1] [INFO] [1721823866.120203156] [foxglove_bridge_component_manager]: Instantiate class: rclcpp_components::NodeFactoryTemplate<foxglove_bridge::FoxgloveBridge>
[foxglove_bridge-1] [INFO] [1721823866.125399664] [foxglove_bridge]: Starting foxglove_bridge (humble, 0.7.7@) with WebSocket++/0.8.2
[foxglove_bridge-1] [INFO] [1721823866.126065432] [foxglove_bridge]: [WS] Server running without TLS
[foxglove_bridge-1] [INFO] [1721823866.126504308] [foxglove_bridge]: [WS] WebSocket server listening at ws://0.0.0.0:8765
[foxglove_bridge-1] [INFO] [1721823866.126538567] [foxglove_bridge]: [WS] WebSocket server run loop started
```
Therefore I can visualize things as expected in Foxglove Studio in the PC. 

2°) On the other hand, I created the following launch file, which include the equivalent launch and other ROS2 functions:
```
from launch import LaunchDescription
from launch.actions import IncludeLaunchDescription, ExecuteProcess
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch_xml.launch_description_sources import XMLLaunchDescriptionSource
from launch_ros.actions import Node
import os

def generate_launch_description():
    # Include rosbridge websocket [launch file] :
    rosbridge_launch_path = os.path.join(get_package_share_directory('rosbridge_server'),
                                     'launch',
                                     'rosbridge_websocket_launch.xml')
    rosbridge_launch = IncludeLaunchDescription( XMLLaunchDescriptionSource(rosbridge_launch_path) )

    # Include foxglove bridge [launch file] :
    foxglove_bridge_launch_path = os.path.join(get_package_share_directory('foxglove_bridge'),
                                           'launch',
                                           'foxglove_bridge_launch.xml')
    foxglove_bridge_launch = IncludeLaunchDescription( XMLLaunchDescriptionSource(foxglove_bridge_launch_path) )

    # Include xsarm moveit [launch file] :
    xsarm_moveit_launch_path = os.path.join(get_package_share_directory('amlab_xsarm_moveit'),
                                            'launch',
                                            'xsarm_moveit.launch.py')
    xsarm_moveit_launch = IncludeLaunchDescription( PythonLaunchDescriptionSource(xsarm_moveit_launch_path),
        launch_arguments={'robot_model': 'wx250s', 'hardware_type': 'actual'}.items() )

    # Include ros2_service_wrapper.py [executable] :
    ros2_service_wrapper = ExecuteProcess( cmd=[ 'python3',
        os.path.join(get_package_share_directory('amlab_xsarm_moveit'), 'ros2_service_wrapper.py') ],
        output='screen')

    # Launch the whole :
    return LaunchDescription( [rosbridge_launch, foxglove_bridge_launch, xsarm_moveit_launch, ros2_service_wrapper] )

# Function to get packages paths :
def get_package_share_directory(package_name):
    from ament_index_python.packages import get_package_share_directory
    return get_package_share_directory(package_name)
```
In this way, the default address and port are not met as can be seen in the terminal feedback :
`[foxglove_bridge-3] [INFO] [1721824501.660898719] [foxglove_bridge]: [WS] WebSocket server listening at ws://[::1]:9090`
It should be `ws://0.0.0.0:8765` instead of `ws://[::1]:9090`), so of course Foxglove Studio cannot get the data :(

3°) I can solve this by specifying the address and the port as launch arguments in my launch file:
```
foxglove_bridge_launch = IncludeLaunchDescription( XMLLaunchDescriptionSource(foxglove_bridge_launch_path),
        launch_arguments={'port': '8765', 'address': '0.0.0.0'}.items() )
```
By doing so I obtain `ws://0.0.0.0:8765` as expected and Foxglove Studio works.

**Expected Behavior**
The result obtained at step 3°) should appear at the step 2°), since **the port and the address are set by default to "8765" and "0.0.0.0"** in `foxglove_bridge_launch.xml`, which is the file called by my launch code. What is the reason for this error?

achim-k, 2024-07-29:
It seems like it picks up the default port of the rosbridge_server, which is [9090](https://github.com/RobotWebTools/rosbridge_suite/blob/d22f102b59e7d9fdeea0ec5e74aa8b98358585d7/rosbridge_server/launch/rosbridge_websocket_launch.xml#L2)

To me this looks like a ROS bug. Both `rosbridge_server` and `foxglove_bridge` launch files have a `port` argument with different default ports and apparently something is going wrong when including both of these launch files.

I assume it correctly uses port `8765` if you do remove `rosbridge_server` from your launch file? Probably changing the order of the two includes will also affect the assigned ports?

qcharmoillau, 2024-07-31:
@achim-k, indeed, the default foxglove bridge port is correct when removing the launching of `rosbridge_server`. It is also correct when `foxglove_bridge` is launched before `rosbridge_launch`, that's to say by using 
`return LaunchDescription([foxglove_bridge_launch, rosbridge_launch, xsarm_moveit_launch, ros2_service_wrapper])` 
instead of 
`return LaunchDescription([rosbridge_launch, foxglove_bridge_launch, xsarm_moveit_launch, ros2_service_wrapper])`. Thank you for the idea.

... Yet, this behavior looks abnormal and there is probably a ROS bug as you said.

achim-k, 2024-07-31:
Feel free to open an issue at https://github.com/ros2/launch. I'm going to close this issue here as the cause for this issue is not foxglove_bridge related.

closed 2024-07-31 by achim-k