foxglove/ros-foxglove-bridge #323: Changing of gradient and min-max scaling value doesn't change the visualization result

TharitSinsunthorn, 2024-09-02:
**Description**
I have set the panel for visualizing the depth camera topic from Isaacsim, using ros2 foxglove bridge. However, when I tried to change the gradient color and also scaling value range, there is no change in the visualization panel.

[image]

**Expectation**
I also tested in rviz and the expected result should show the image from depth camera more clear

[image]

Here is the reference from RGB image

[image]

achim-k, 2024-09-02:
This issue does not seem related to foxglove bridge but rather to the Foxglove app. Could you open the bug report either at https://github.com/orgs/foxglove/discussions or ask for support on [Discord](https://foxglove.dev/chat)? I'm going to close this issue.

closed 2024-09-02 by linear[bot]

TharitSinsunthorn, 2024-09-02:
OK, sorry for confusing