foxglove/ros-foxglove-bridge #330: Errors building foxglove-bridge 0.7.7-1 with Yocto (kirkstone) and ros2 humble

grahas, 2024-12-11:
**Description**
When building with yocto the following error is seen

`
/home/axcend/OSTL-k/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-oe-linux-gnueabi/foxglove-bridge/0.7.7-1-r0/git/ros2_foxglove_bridge/src/ros2_foxglove_bridge.cpp:482:50: error: no matching function for call to 'max(long unsigned int, size_t)'
  482 |     const size_t publisherHistoryDepth = std::max(1ul, qos.depth());
      |                                          ~~~~~~~~^~~~~~~~~~~~~~~~~~
`

- Version: 0.7.7-1
- Platform: target is arm

**Steps To Reproduce**

bitbake foxglove-bridge

**Expected Behavior**
no build error

jtbandes, 2024-12-11:
Thanks – I believe you reported the same issue in Discord as well. If you verify the fix works for you then please submit a PR and we can review+merge.

defunctzombie, 2024-12-13:
Fixed in https://github.com/foxglove/ros-foxglove-bridge/pull/331

closed 2024-12-13 by defunctzombie