foxglove/ros-foxglove-bridge #333: Service request with multiple array fields seems to fail

MCFurry, 2024-12-30:
**Description**

For example, when using foxglove studio to connect to foxglove-bridge, the following service call on `my_node/set_parameters` succeeds:
`{
  "parameters": [
    {
      "name": "a",
      "value": {
        "type": 3,
        "double_value": 1.0
      }
    }
  ]
}`

But once the array contains more than one element, it does not. I.e.:
`{
  "parameters": [
    {
      "name": "a",
      "value": {
        "type": 3,
        "double_value": 1.0
      }
    },
    {
      "name": "b",
      "value": {
        "type": 3,
        "double_value": 2.0
      }
    }
  ]
}`

Makes the node crash with the exception: 

> terminate called after throwing an instance of 'rclcpp::exceptions::InvalidParametersException'
  what():  parameter name must not be empty

While the ROS2 CLI equivalent works correctly:
`ros2 service call /my_node/set_parameters rcl_interfaces/srv/SetParameters "{"parameters": [{"name": "a","value": {"type": 3,"double_value": 1.0}},{"name": "b","value": {"type": 3,"double_value": 2.0}}]}"`

When building foxglove_bridge from source and [adding some debug prints here](https://github.com/foxglove/ros-foxglove-bridge/compare/main...MCFurry:ros-foxglove-bridge:debug-service-requests) it seems the service request is not received correctly as I get (note the second entry is empty/default):

> [INFO] [1735558998.294195957] [foxglove_bridge]: Calling service: /my_node/set_parameters with content: 
[INFO] [1735558998.294419992] [foxglove_bridge]: Parameter: a
[INFO] [1735558998.294451375] [foxglove_bridge]: Value: 1.000000
[INFO] [1735558998.294462719] [foxglove_bridge]: Parameter: 
[INFO] [1735558998.294488947] [foxglove_bridge]: Value: 0.000000

- Version: 0.8.2
- Platform: Linux Ubuntu 24.04

**Steps To Reproduce**
Simply do a service call from foxglove studio to any node's `/set_parameters` service with more than 1 entry in the parameters list.

**Expected Behavior**
I expect the service call simply goes through unchanged and thereby applies the two (or more) requested parameter changes.

**Note**
It might very well be that the actual bug is inside foxglove studio. But since that's closed source nowadays, I cannot easily isolate who is to blame or where stuff goes wrong...

achim-k, 2024-12-30:
I believe that you have to include all `xy_value` fields, not just `double_value` for each parameter. I can successfully set multiple parameters with Foxglove's service call panel when specifying all fields. This does not work when only specifying the `double_value` field.

[image]

MCFurry, 2024-12-30:
Thanks for your swift reply!
I see, that indeed seems to fix the issue.

But isn't it strange that only `double_value` does work for a single parameter but not for more?

And also works with standard ROS2 cli tools?
i.e. `ros2 service call /my_node/set_parameters rcl_interfaces/srv/SetParameters "{"parameters": [{"name": "a","value": {"type": 3,"double_value": 1.0}},{"name": "b","value": {"type": 3,"double_value": 2.0}}]}"`
I'd expected the raw json to simply be forwarded to a ROS service call?

achim-k, 2024-12-31:
Turns out that Foxglove incorrectly serialized the request message when array fields were missing. After applying the fix locally I can set multiple parameters without providing the non-required value fields. I will provide a proper fix for this next year :wink:  (happy new year btw)

[image]

MCFurry, 2025-01-02:
Quick  find, thanks! 
Happy new year to you too!

MCFurry, 2025-01-06:
If you have any change implemented that I can test, please let me know and we can close this issue!

achim-k, 2025-01-07:
The actual bug fix has been made in https://github.com/foxglove/rosmsg2-serialization/pull/37. This fix will be included in the next Foxglove release and potentially already today in the web version. I will close this issue once the internal PR gets merged.

closed 2025-01-08 by achim-k