foxglove/ros-foxglove-bridge #335: Incorrect Parameter Type

pitosalas, 2025-01-24:
I am including the foxglove launch file in my yaml launch file:

```
launch:
  - include:
      file: "$(find-pkg-share linorobot2_bringup)/launch/bringup.launch.py"
      arg:
        - name: base_serial_port
          value: "/dev/ttyUSB0"
        - name: micro_ros_baudrate
          value: "921600"
  - include:
      file: "$(find-pkg-share linorobot2_description)/launch/description.launch.py"
  - include:
      file: "$(find-pkg-share hls_lfcd_lds_driver)/launch/hlds_laser.launch.py"
  - include:
      file: "$(find-pkg-share foxglove_bridge)/launch/foxglove_bridge_launch.xml"

```

And there is some kind of parse error buried inside your launch config:

[foxglove_bridge-8]   what():  parameter 'port' has invalid type: Wrong parameter type, parameter {port} is of type {integer}, setting it to {string} is not allowed.

jtbandes, 2025-01-24:
Can you share more details about your setup, like what ROS version are you using?

pitosalas, 2025-01-25:
Ros2 humble. 

I fixed it by overriding that argument by hand. So no big deal. But it is a tiny bug I think!

Thanks

achim-k, 2025-01-31:
What ROS version and what foxglove_bridge version are you using?

On ROS iron with foxglove_bridge v0.8.1 I can launch the following launch files without any issues:

```yaml
launch:
  - include:
      file: "$(find-pkg-share foxglove_bridge)/launch/foxglove_bridge_launch.xml"
```

achim-k, 2025-02-11:
Closing this issue as it cannot be reproduced and no further details have been provided.

closed 2025-02-11 by achim-k