foxglove/ros-foxglove-bridge #337: foxglove_bridge: Exception when getting parameters

pitosalas, 2025-02-14:
I am getting this error:

[foxglove_bridge-7] [ERROR] [1739562014.017118712] [foxglove_bridge]: Exception when getting parameters: Failed to retrieve parameter names for node '/hlds_laser_publisher'

As far as I can see, hlds_laser_publisher is not publishing any parameter values. I don' know why foxglove is asking.

amacneil, 2025-02-14:
Hi, thanks for reporting!

Can you please provide a minimal git repo (and/or docker image) to reproduce the issue?

pitosalas, 2025-02-16:
I am using this repo https://github.com/ROBOTIS-GIT/hls_lfcd_lds_driver/blob/main/src/hlds_laser_publisher.cpp

But I just saw the error again: 
[foxglove_bridge-7] [ERROR] [1739716232.027241021] [foxglove_bridge]: Exception when getting parameters: Failed to retrieve parameter names for node '/velocity_controller'

I don't know if that's a known issue with humble and foxglove or maybe I should be using rosbridge or one of the other protocols.

achim-k, 2025-02-28:
I'm going to close this issue. @pitosalas feel free to re-open if you are still facing this issue.

closed 2025-02-28 by achim-k

dkozma, 2025-03-25:
@achim-k I'm also running into this issue, and was wondering if there was a way to deny param checking for certain nodes since it would be good to not eliminate parameter information across the board via the `capabilities` parameter. I'm getting this for the [`/SlamToolboxPlugin`](https://github.com/SteveMacenski/slam_toolbox/tree/ros2/rviz_plugin) (in addition to some of my own plugins) that don't use params.

When I run `ros2 param list /SlamToolboxPlugin` (or run `param list` on any nodes that are failing param handling in Foxglove) it hangs, so it's a legitimate issue, however it would be great if I could still use Foxglove normally if this happens. I'm still quite new to ROS2 so if anyone has a way to fix the hanging issue, that would also be appreciated.