foxglove/ros-foxglove-bridge #338: Channels changing

pitosalas, 2025-02-16:
These messages happen one after the other, in a row. Is that normal or is it worth looking into?

```
[foxglove_bridge-7] [INFO] [1739715857.147431619] [foxglove_bridge]: Client 192.168.5.76:62217 is advertising "/cmd_vel" (geometry_msgs/msg/Twist) on channel 1 with encoding "cdr"
[foxglove_bridge-7] [INFO] [1739715857.147702433] [foxglove_bridge]: Client 192.168.5.76:62217 is no longer advertising /cmd_vel (1 subscribers) on channel 1
[foxglove_bridge-7] [INFO] [1739715857.153217832] [foxglove_bridge]: Client 192.168.5.76:62217 is advertising "/cmd_vel" (geometry_msgs/msg/Twist) on channel 2 with encoding "cdr"
[foxglove_bridge-7] [INFO] [1739715857.153469127] [foxglove_bridge]: Client 192.168.5.76:62217 is no longer advertising /cmd_vel (1 subscribers) on channel 2
[foxglove_bridge-7] [INFO] [1739715857.159007934] [foxglove_bridge]: Client 192.168.5.76:62217 is advertising "/cmd_vel" (geometry_msgs/msg/Twist) on channel 3 with encoding "cdr"
[foxglove_bridge-7] [INFO] [1739715857.159256377] [foxglove_bridge]: Client 192.168.5.76:62217 is no longer advertising /cmd_vel (1 subscribers) on channel 3
[foxglove_bridge-7] [INFO] [1739715857.166125603] [foxglove_bridge]: Client 192.168.5.76:62217 is advertising "/cmd_vel" (geometry_msgs/msg/Twist) on channel 4 with encoding "cdr"
```

achim-k, 2025-02-17:
Does this happen when you connect Foxglove to the bridge? Do you use a custom panel that advertises the `/cmd_vel` topic?

defunctzombie, 2025-03-21:
Closing for inactivity. We can take a look if more info is provided.

closed 2025-03-21 by defunctzombie